micropython/micropython #6195: STM32H7 Ethernet

rcordeau, 2020-06-26:
Hi,
Likely this is a feature request - but is it feasible to get ethernet working for the H7 series of STM32 chips? I know the registers are different compared to the other chips, but I wasn't sure if anyone was looking in to this.

Thanks,
Rob

dpgeorge, 2020-06-28:
> Likely this is a feature request

Right, it's currently not implemented.  It only needs the very low-level ETH interface on the H7 to be implemented, everything else (TCP/IP stack and upwards) is already done.  I might have some time to look into it.

rcordeau, 2020-06-28:
I was going to update it for the H7 anyway for my board. Should I do that and then put up a pull request, or would you rather do it? 

Rob

dpgeorge, 2020-06-28:
> Should I do that and then put up a pull request,

Yes that would be great!

rf-eng, 2020-07-15:
Hi,
I adapted the code in eth.c and tested it on a H743 Nucleo board. My changes can be found here: https://github.com/rf-eng/micropython/commit/e2c24910c03bd4f786d67ca4991b31df99a75044.
At least pinging seems to work reliably, but I am not sure how this should be included into the existing code. Using a large `#ifdef` section? Maybe someone can have a look and give me a hint before I create a pull request. There are quite a few changes compared to the original implementation, because of different registers, a different structure of the ethernet descriptors and the requirement for H7 to put descriptors/buffers into D2 RAM.

rcordeau, 2020-07-15:
> I adapted the code in eth.c and tested it on a H743 Nucleo board. My changes can be found here: rf-eng@e2c2491 <https://github.com/rf-eng/micropython/commit/e2c24910c03bd4f786d67ca4991b31df99a75044>.

Sweet! Thank you Reinhard!

Wian144, 2022-02-03:
Hi all.

I am not sure if this request warrants a separate ticket, if it does I will gladly move it elsewhere. 

I have project where I need to connect the STM32H743 to a MAC through MII. Has anyone done this before? I would appreciate any pointers on where to change the HAL source code to enable this feature. I believe it should be as simple as configuring a few registers and assigning the correct pins.

Any help would be appreciated.

Thanks, 
Wian.

rolandvs, 2022-02-03:
The nucleo_h743 board supports this

dpgeorge, 2022-02-03:
> AFAIK the Micropython port for the Nucleo-H743 only supports RMII. I know the board does support MII, but would like to add functionality for it in Micropython.

I have not tried it, but I think all that is required is:
1. configure the GPIO correctly for MII instead of RMII, in `eth.c:eth_init()`
2. select MII instead of RMII in `SYSCFG->PMCR`, in `eth.c:eth_mac_init()`

rolandvs, 2022-02-03:
Just to be sure, I did not check this yet, but rmii uses 50MHz clock and mii probably 25MHz. So maybe you need to verify that too...