micropython/micropython #6198: Importing a function into class context does not work properly

aivarannamaa, 2020-06-26:
Following code runs fine in Python 3.7:

```
class C:
    from os import listdir

print(C.listdir("/"))
```

but fails in MicroPython 1.12 on ESP 32:

```
>>> 
paste mode; Ctrl-C to cancel, Ctrl-D to finish
=== class C:
===     from os import listdir
=== 
=== print(C.listdir("/"))
=== 
Traceback (most recent call last):
  File "<stdin>", line 5, in <module>
TypeError: argument has wrong type
```

The method itself is present and well:

```
>>> from os import listdir
>>> listdir("/")
['boot.py', 'webrepl_cfg.py', 'main.py', 'pass26_crop.jpg']
```

Curiously, no argument functions are imported into class context properly:

```
paste mode; Ctrl-C to cancel, Ctrl-D to finish
=== class C:
===     from os import getcwd
=== 
=== print(C.getcwd())
=== 
/
```

aivarannamaa, 2020-06-26:
Error message from the Unix port implies, that the imported function is converted to something like an instance method:

```
paste mode; Ctrl-C to cancel, Ctrl-D to finish
=== class C:
===     from os import ilistdir
=== 
=== print(C.ilistdir("/"))
=== 
=== 
Traceback (most recent call last):
  File "<stdin>", line 4, in <module>
TypeError: argument should be a 'C' not a 'str'
```

aivarannamaa, 2020-06-26:
It looks like the problem is more general -- here is my attempt to assign imported function into a class field:

```
paste mode; Ctrl-C to cancel, Ctrl-D to finish
=== class C:
===     pass
=== 
=== import os
=== C.listdir = os.listdir
=== print(C.listdir("/"))
=== 
=== 
Traceback (most recent call last):
  File "<stdin>", line 6, in <module>
TypeError: argument has wrong type
```

peterhinch, 2020-06-27:
I can't replicate this:
```python
MicroPython v1.12-535-g509359754-dirty on 2020-06-14; linux version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> 
paste mode; Ctrl-C to cancel, Ctrl-D to finish
=== class C:
===     pass
=== 
=== import os
=== C.listdir = os.listdir
=== 
=== print(C.listdir('/'))
=== 
=== 
['cdrom', 'media', 'lib64', 'proc', 'run', 'opt', 'var', 'target', 'home', 'lost+found', 'initrd.img', 'root', 'bin', 'sbin', 'boot', 'vmlinuz', 'initrd.img.old', 'tmp', 'srv', 'lib', 'mnt', 'vmlinuz.old', 'sys', 'dev', 'swapfile', 'usr', 'snap', 'etc']
>>> 
```

dpgeorge, 2020-06-27:
Thanks @aivarannamaa for the report.  I can reproduce the problem.  It's related to #1326 but here the native method is looked up in the class not the instance (eg `C.listdir` vs `C().listdir`).  These two issues should be fixed together.

@peterhinch it looks like you are importing `os.py` which is a Python wrapper for the built-in `uos` and so `os.listdir` is a Python function that doesn't have the issue.

dpgeorge, 2020-07-01:
Fixed by 332d83343fb3ef5d2b94b4f058aa53fd0493779e

closed 2020-07-01 by dpgeorge