micropython/micropython #6203: import this Not working

mrkeuz, 2020-06-28:
MicroPython v1.12-590-g9f911d822 on 2020-06-28; linux version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> import this
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: no module named 'this'
>>>

stinos, 2020-06-28:
MicroPython is meant to be 'micro' so there's no compelling reason to make it larger by including arbitrary text which has limited use.

Is there any practical reason you need this? You can always just download https://github.com/python/cpython/blob/master/Lib/this.py and put it in you search path, so you can `import this`.

mrkeuz, 2020-06-28:
ok. Now I understend the 'micro' concept of micropython. Just in my mind i have expectaction of fully featured python. Also "import this" is cool branded easter egg of python and it would be nice to have it in core.

But I argee with you point ~1000 bytes of this module will make sense for some tiny ports.

closed 2020-06-28 by mrkeuz