micropython/micropython #6204: How to Compiling uPy on Windows 7/10?

GianCann, 2020-06-28:
Hi to all,

I cloned the repository and I installed MinGW with this package:
[image]

When I try to execute make, I get this error:
```
C:\Users\remoteadmin\micropython\mpy-cross>make
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosit
y.
process_begin: CreateProcess(NULL, uname -s, ...) failed.
mkdir -p build/genhdr
Sintassi del comando errata.
../py/mkrules.mk:121: recipe for target 'build/genhdr' failed
make: *** [build/genhdr] Error 1
```

Thank you for any help.

stinos, 2020-06-28:
```
mkdir -p build/genhdr
Sintassi del comando errata.
```

You're running this from a command prompt, that's not really supported: the Makefile expects to be ran in a unix-like environment, i.e. using NU mkir. What happens here though is calling mkdir.exe with a syntax which is incorrect. In short: start an msys2/bash shell (I have no experience with the GUI you show so I cannot tell how to do that exactly) and run make from there.

jonnor, 2025-10-03:
There are instructions at https://github.com/micropython/micropython/tree/master/ports/windows for how to build to run on Windows.

In general however using Windows Subsystem for Linux (WSL) might be better supported in most cases, for use on hardware.

closed 2025-10-03 by jonnor