micropython/micropython #6209: docs: Escaping `*` in argument list no longer necessary

jimmo, 2020-06-29:
Compare http://docs.micropython.org/en/v1.12/library/pyb.html#pyb.mount and http://docs.micropython.org/en/latest/library/pyb.html#pyb.mount

`latest` is rendered as `\*`, whereas `1.12` is `*`.

An easy fix, but would be good to get to the bottom of why this changed.

jimmo, 2020-06-29:
Was raised in the forum at https://forum.micropython.org/viewtopic.php?f=2&t=8584&p=48674

hlovatt, 2020-06-29:
It’s not just mount, it’s throughout the documentation!

dpgeorge, 2020-06-29:
I guess it's because Sphinx was upgraded to 3.1.0.

Zenix27, 2020-07-10:
I would like to work on this issue #6209, just wanted a heads-up if `\*` has to be replaced with`*` in the entire documentation?

hlovatt, 2020-07-10:
It needs replacing in argument list throughout the `.rst` files I think. I haven’t looked at all files by any means, but it is true in all I have looked at. I don’t understand the mark up well enough to know if * (without escaping) is OK in main body or just argument list. 

-- Howard.

Zenix27, 2020-07-11:
I have raised a PR, went through all the `.rst` files in micropython/docs/library/ 
@jimmo @dpgeorge

Zenix27, 2020-07-12:
there were problems with space in some commits, now resolved - #6239

dpgeorge, 2020-08-21:
Fixed by e76c7466b62c940150cf1b5000e18f60a5efa2e3

closed 2020-08-21 by dpgeorge