micropython/micropython #6216: Help for W5500 and Display

Benni08, 2020-07-01:
Hey everybody,

after hours of Searching, i can't help myself, so i have to ask.

I Have a m5Stack with a offical Ethernet Shield (underlying W5500 Chip). But i can't find any firmware to flash, wich supports the W5500 Chip and also have drivers for the LCD.

So as i See, i have to build it on my own.

Installing the Firmware from offical micropython Source, don't get me the Drivers for W5500 and Display. So as i've readed in the Wiki, i need to 

> The particular chipset that is supported by the firmware is selected at compile-time via the MICROPY_PY_WIZNET5K option

But i don't know how.

Best Regards and be aware of my bad english.

Ben

hu-lu, 2020-07-04:
When I added w5500 support, I made two changes：
one: add “#define MICROPY_PY_NEWORK        (1)” to mpconfigboard.h
two: Change "MICROPY_PY_WIZNET5K ?= 0" to "MICROPY_PY_WIZNET5K ?= 5500" in stm32/mpconfigport.mk
Then recompile the source code.

jonnor, 2024-07-03:
There seems to be a fix for this issue provided already. Proposing to close this.

projectgus, 2024-07-05:
Yes, I think either it should be possible to build with the MicroPython driver using this comment, or this issue can be rolled into #8537.

closed 2024-07-05 by projectgus