micropython/micropython #6219: [ESP32] Read from ADC2 Pins

jdjjm, 2020-07-02:
Is there any way of accessing ADC2 Pins? 
Even with WiFi off you cannot set up an ADC on any of the ADC2 Pins as you get the error
`ValueError: invalid Pin for ADC`

Looking at the esp32 port for [machine_adc.c](https://github.com/micropython/micropython/blob/master/ports/esp32/machine_adc.c) it seems only adc1 pins are supported. Is there any way to use ADC2?

peterhinch, 2020-07-02:
Please raise this in [the forum](http://forum.micropython.org/index.php) rather than in the MicroPython bug tracker.

jdjjm, 2020-07-02:
Okay thanks

jdjjm, 2020-07-02:
Is there a reason [machine_adc](https://github.com/micropython/micropython/blob/master/ports/esp32/machine_adc.c) only exposes ADC 1? I understand most uses case will have WiFi running however there are times when this isn't the case so what is the reason for not having them accessible? It seems a little restrictive to me.

WayneKeenan, 2020-07-02:
FYI: There is a workaround for getting ADC2 values whilst WiFi is running. 
Also, an Espressif dev said (albeit a long time ago) "There is a fix coming for ADC2 and WiFi soon :)"

Ref:  https://github.com/espressif/arduino-esp32/issues/102#issuecomment-331360699

dpgeorge, 2020-07-02:
This is simply not implemented (mainly because with WiFi enabled it's not possible to use ADC2).  Also there's currently no standard way to select an ADC peripheral to use for a given pin, but in this case I think it could just automatically select ADC2 if a pin was given that only supported ADC on ADC2.

Summary: it can be implemented, feel free to make a PR for it.

dpgeorge, 2022-01-21:
Support for ADC2 was implemented in 63438a31bb6d0c8012e338433df95700f1cbca97

closed 2022-01-21 by dpgeorge

MauroDiamantino, 2022-01-28:
> Support for ADC2 was implemented in [63438a3](https://github.com/micropython/micropython/commit/63438a31bb6d0c8012e338433df95700f1cbca97)

But, do I have to disable Wifi to use it?

dpgeorge, 2022-02-02:
> But, do I have to disable Wifi to use it?

I'm not sure, you'll need to read the ESP datasheet, or just try it out.

commit 63438a31 (2022-01-21): esp32/machine_adcblock: Add new machine.ADCBlock class and update ADC.

Rework the ADC implementation to follow the improved ADC/ADCBlock API.
This adds support for calibrated voltage readings and the ADC2 block.  The
ADC API is backwards compatible with what it was before this change.

Resolves #6219.