micropython/micropython #6225: argument count error is wrong when subclass calls super()

dhalbert, 2020-07-03:
A CircuitPython user noticed this error (https://github.com/adafruit/circuitpython/issues/3112). It is also present in the latest build of MicroPython. An incorrect number of arguments is reported in the exception when a subclass constructor is given the wrong number of arguments, and that constructor calls the superclass constructor.
```python
class A:
    def __init__(self):
        pass

class B(A):
    def __init__(self):
        super().__init__()  # if this is replaced with pass, the problem goes away

try:
    print("A(1)")
    A(1)
except Exception as e:
    print(e)

try:
    print("B(1)")
    B(1)
except Exception as e:
    print(e)
```

```python
MicroPython v1.12-614-gc2317a3a8 on 2020-07-03; PYBv1.1 with STM32F405RG
Type "help()" for more information.
>>> import args
A(1)
function takes 1 positional arguments but 2 were given
B(1)
function takes 2 positional arguments but 3 were given      # should be 1 and 2
```

jimmo, 2020-07-04:
The issue is that the `super()` is implicitly converted to `super(__class__, self)`, but somehow referencing `__class__` from the parent scope is causing the arg count calculation to get messed up. Here's a simpler repro without `__init__` or `super()`:

```py
class A:
    def x(self):
        print(__class__, self)

try:
    A().x(1)
except Exception as e:
    print(e)
```

```txt
x() takes 2 positional arguments but 3 were given
```

dpgeorge, 2020-07-08:
When a class function accesses `__class__` then it treats `__class__` as a closed-over variable and creates a closure (eg the function `x()` above is a closure).  And this closure of `__class__` is shared with the class body (the code that builds the class).  There's special code to handle all this and because `x()` takes an extra hidden argument (the closure cell) that's counted as an argument when the error is printed.

I'm not sure why that closure is needed... I'd need to review the code in detail, but it's possible it could be replaced with a simple lookup in the class dict.

dpgeorge, 2020-07-08:
> I'm not sure why that closure is needed

I guess it's because functions defined within a class scope have no way of knowing their `__class__` and it's not possible to inspect the `self` object because that would then retrieve the wrong class in the case of a inherited class (eg a method in class `A` would retrieve class `B`).