micropython/micropython #6226: Unix port can't run readable files belonging to another user

aivarannamaa, 2020-07-04:
```
annamaa:~/python_stuff$ echo "print('hi')" > test.py
annamaa:~/python_stuff$ micropython test.py 
hi
annamaa:~/python_stuff$ ls -l test.py 
-rw-r--r-- 1 annamaa annamaa 12 juuli  4 07:49 test.py
annamaa:~/python_stuff$ sudo chown demo test.py
annamaa:~/python_stuff$ micropython test.py 
OSError: [Errno 13] EACCES
annamaa:~/python_stuff$ python3 test.py 
hi
```

aivarannamaa, 2020-07-04:
It looks like it also can't read them:

```
MicroPython 15fd8a4 on 2019-12-22; linux version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> open("test.py")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 13] EACCES
```

stinos, 2020-07-04:
I assume this is the culprit, in vfs_posix_file?

```
    open(fname, mode_x | mode_rw, 0644)
```

dpgeorge, 2020-07-08:
I can't reproduce this.  Repeating the above on Arch Linux it works ok, `micropython` can execute and open for reading such a file.

aivarannamaa, 2020-07-20:
I made this test with `micropython` installed with `snap` and forgot about its sandboxing thing. 

When I now built MicroPython from source myself, everything worked fine.

closed 2020-07-20 by aivarannamaa