micropython/micropython #6227: usocket: Support context manager

jimmo, 2020-07-04:
Raised in the forum: https://forum.micropython.org/viewtopic.php?f=15&t=8641

Would this require anything else other than adding `__enter__` and `__exit__` to the various socket implementations (`esp32/modsocket.c`, `extmod/modlwip.c`, `unix/modusocket.c`, `stm32/modusocket.c`).

Likely not a good idea, but the other alternative would be to make the `with` statement able to detect that the object implements the stream protocol...? Would get complicated supporting that in the native emitter too.

dpgeorge, 2020-07-08:
I'd guess `__enter__` is a no-op and `__exit__` simply does a close on the socket.  It might then be possible to do it very cheaply by linking `__enter__` to `&mp_identity_obj` and then have a common `__exit__` method that's shared among these kind of stream objects (eg files).

> but the other alternative would be to make the `with` statement able to detect that the object implements the stream protocol...?

Could do.  Or such types could have a bit set in their `type.flags` which indicates they have a no-op/close context manager and the runtime then has special code to handle `__enter__` and `__exit__`.  Probably it's cleaner to just use the above approach.

Potaten2015, 2021-11-30:
Hey all, just checking if there is any update on this. It would keep things a little bit cleaner on my micropython server. Thanks!

Hereux, 2023-09-05:
Yeah would be nice if @dpgeorge anyone could give us some details.