micropython/micropython #6230: Compiling  Micropython v1.12 in Cygwin in Windows 10

unnamalai-kb, 2020-07-07:
Using Cygwin64 in Windows 10, I tried to build for stm32 and the default board. 

cd micropython
make -C mpy-cross
cd ports/stm32
make submodules
export PATH="/cygdrive/c/gnu_arm_embedded/bin:$PATH"
make V=1

But the compilation always ends with the following error in **BOLD**:
..
..
arm-none-eabi-gcc -I. -I../.. -Ibuild-PYBV10 -I../../lib/cmsis/inc -I../../lib/stm32lib/CMSIS/STM32F4xx/Include/ -I../../lib/stm32lib/STM32F4xx_HAL_Driver/Inc -Iusbdev/core/inc -Iusbdev/class/inc -Ilwip_inc -Wall -Wpointer-arith -Werror -std=gnu99 -nostdlib -DFFCONF_H=\"lib/oofatfs/ffconf.h\" -DMICROPY_VFS_FAT=1 -DMICROPY_VFS_LFS2=1 -DLFS2_NO_MALLOC -DLFS2_NO_DEBUG -DLFS2_NO_WARN -DLFS2_NO_ERROR -DLFS2_NO_ASSERT  -DSTM32F405xx -mthumb -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mtune=cortex-m4 -mcpu=cortex-m4 -Os -DNDEBUG -Iboards/PYBV10 -DSTM32_HAL_H='<stm32f4xx_hal.h>' -DMICROPY_HW_VTOR=0x08000000 -DMICROPY_FLOAT_IMPL=MICROPY_FLOAT_IMPL_FLOAT -fsingle-precision-constant -Wdouble-promotion -fdata-sections -ffunction-sections -DMICROPY_MODULE_FROZEN_STR -DMICROPY_QSTR_EXTRA_POOL=mp_qstr_frozen_const_pool -DMICROPY_MODULE_FROZEN_MPY -c -MD -o build-PYBV10/lib/utils/printf.o ../../lib/utils/printf.c
python3 ../../tools/makemanifest.py -o build-PYBV10/frozen_content.c -v "MPY_DIR=../.." -v "MPY_LIB_DIR=../../../micropython-lib" -v "PORT_DIR=/cygdrive/c/Bridgetek/01_Project/25_Micropython/01_work/git/micropython/ports/stm32" -v "BOARD_DIR=boards/PYBV10" -b "build-PYBV10" -f"-march=armv7m" boards/manifest.py
**mpy-cross not found at C:\Bridgetek\01_Project\25_Micropython\01_work\git\microp
ython/mpy-cross/mpy-cross, please build it first**
**make: *** [../../py/mkrules.mk:103: build-PYBV10/frozen_content.c] Error 1**

I have tried clean building mpy-cross, but I get the same error for stm32 build. 

I tried the following in the mpy-cross folder. 
$ ./mpy-cross.exe --version
MicroPython v1.12 on 2020-07-07; mpy-cross emitting mpy v5

So mpy-cross.exe is available. Please help with the error above in building stm32.

stinos, 2020-07-07:
See line 61 in mkenv.mk: the port expects to find `mpy-cross` but you have `mpy-cross.exe`.
Passing something like `MPY_CROSS=../../mpy-cross/mpy-cross.exe` to `make` should fix that.

unnamalai-kb, 2020-07-07:
Thank you. In addition to mkenv.mk, I had to make changes to the file  micropython\tools\manifest.py line no 219 to,    
MPY_CROSS = VARS['MPY_DIR'] + '/mpy-cross/mpy-cross.exe'
It can now build the firmware binary.

stinos, 2020-07-07:
That should be fixed in recent versions though, since the code there is now
```
    MPY_CROSS = VARS["MPY_DIR"] + "/mpy-cross/mpy-cross"
    if sys.platform == "win32":
        MPY_CROSS += ".exe"
```

unnamalai-kb, 2020-07-07:
Thanks, I will take a look in the recent version. I'll close the issue now.

closed 2020-07-07 by unnamalai-kb