micropython/micropython #6231: LEDs not working on STM32F4DISCOVERY

mb, 2020-07-07:
Even though the STM32F4DISCOVERY has `GPIOD` pins, they don't get initialized in `stm32_main`. 
https://github.com/micropython/micropython/blob/c2317a3a8d5f184de2f816078d91be699274b94e/ports/stm32/main.c#L452-L454

The following program doesn't have any effect on the LEDs, because the user LEDs are connected through `GPIOD`:

```python
>>> led = pyb.LED(1)
>>> led.on()
```

Removing the guards in `ports/stm32/main.c` works as a workaround:

```patch
diff --git a/ports/stm32/main.c b/ports/stm32/main.c
index b5dbfa50f..590756de8 100644
--- a/ports/stm32/main.c
+++ b/ports/stm32/main.c
@@ -449,9 +449,9 @@ void stm32_main(uint32_t reset_mode) {
     __HAL_RCC_GPIOA_CLK_ENABLE();
     __HAL_RCC_GPIOB_CLK_ENABLE();
     __HAL_RCC_GPIOC_CLK_ENABLE();
-    #if defined(GPIOD)
     __HAL_RCC_GPIOD_CLK_ENABLE();
-    #endif

     #if defined(STM32F4) || defined(STM32F7)
     #if defined(__HAL_RCC_DTCMRAMEN_CLK_ENABLE)
```

I think including `lib/stm32lib/CMSIS/STM32F4xx/Include/stm32f4xx.h` could fix the problem as GPIOD is defined there.

dpgeorge, 2020-07-08:
When `led_init()` is called it enables the corresponding GPIO clock, so should be enabling GPIOD on the `STM32F4DISC` board.  Also, on this board the line `__HAL_RCC_GPIOD_CLK_ENABLE()` is actually being called because GPIOD is defined.

So, I'm not sure why `pyb.LED(1).on()` does not work.  Might be worth double checking the firmware, board and actual hardware match.

mb, 2020-07-08:
Tried to reproduce it again, but it just worked now. Thanks for the response and sorry for bothering you, the issue was likely on my side. I though, that I double checked it before submitting.. And true `GPIOD` is actually defined in the source code.

closed 2020-07-08 by mb