micropython/micropython #6232: Problem I2C on UART Grove port with Adafruit Feather STM32F405 Express

david-therincourt, 2020-07-08:
Hi,
Is not possible to use the second I2C with a Grove Shield FeatherWing on Feather F405 Express due reverse Grove wire : SCL -> RX and SDA -> TX

Is it possible to invert `pin_B10` and `pin_B11` in `mpconfigboard.h` for I2C busses ?

/micropython/ports/stm32/boards/ADAFRUIT_F405_EXPRESS/mpconfigboard.h

```
// I2C busses
#define MICROPY_HW_I2C1_NAME "I2C1"
#define MICROPY_HW_I2C1_SCL (pin_B6)  // SCL
#define MICROPY_HW_I2C1_SDA (pin_B7)  // SDA
#define MICROPY_HW_I2C2_NAME "I2C2"
#define MICROPY_HW_I2C2_SCL (pin_B10)  // TX
#define MICROPY_HW_I2C2_SDA (pin_B11)  // RX
```

jimmo, 2020-07-08:
Sorry I'm not quite sure what you're asking, what's the actual issue here?

When used in the I2C2 AF mode, PB10 is SCL and PB11 is SDA.

However, if you need to use specific pins, you can use "soft" I2C instead by using

```
i2c = machine.I2C(-1, sda=machine.Pin.board.TX, scl=machine.Pin.board.RX)
```

(Soft I2C can use any pins)

david-therincourt, 2020-07-13:
Sorry for confusion. I remember this is hardware I2C pins. So is only possible to use a soft I2C on UART connector with Grove.
Thanks.

closed 2020-07-13 by david-therincourt