micropython/micropython #6244: Shifting array values right with memoryview clears array

fragmuffin, 2020-07-15:
When shifting values in an `array` +1 index (to the right) using memoryview, 0th index value is carried through the array.

**cPython**

```python
>>> from array import array
>>> a = array('I', range(10))
>>> mv = memoryview(a)
>>> a  # before shifting right
array('I', [0, 1, 2, 3, 4, 5, 6, 7, 8, 9])
>>> mv[1:] = mv[:-1]
>>> a  # after
array('I', [0, 0, 1, 2, 3, 4, 5, 6, 7, 8])  # <-- looks good
```

**micropython**
```python
>>> from array import array

# shift left
>>> a = array('I', range(10))
>>> mv = memoryview(a)
>>> a  # before shifting right
array('I', [0, 1, 2, 3, 4, 5, 6, 7, 8, 9])
>>> mv[1:] = mv[:-1]
>>> a  # after
array('I', [0, 0, 0, 0, 0, 0, 0, 0, 0, 0])  # <-- 0th index value carried through entire array

# shift right
>>> a = array('I', range(10))
>>> mv = memoryview(a)
>>> a  # before shifting left
array('I', [0, 1, 2, 3, 4, 5, 6, 7, 8, 9])
>>> mv[:-1] = mv[1:]
>>> a  # after
array('I', [1, 2, 3, 4, 5, 6, 7, 8, 9, 9])  # <-- looks good
```

fragmuffin, 2020-07-15:
Workaround: populate memoryview from a copy of the array:

```python
>>> mv[1:] = a[:-1]
>>> a
array('I', [0, 0, 1, 2, 3, 4, 5, 6, 7, 8])
```

Or, for a large `array`, manually loop through it backwards:
```python
>>> for i in range(len(a)-1, 0, -1):
...     a[i] = a[i-1]
```

peterhinch, 2020-07-15:
It also happens with a bytearray:
```python
a = bytearray(range(10))
mv = memoryview(a)
mv[1:] = mv[:-1]
print(a)
```

davehylands, 2020-07-15:
This might be a case of using memcpy versus memmove in the C code.

dpgeorge, 2020-07-16:
Thanks @fragmuffin for the report, @peterhinch for the extra test, and @dhylands for the hint about the fix.  Should be fixed by #6249

dpgeorge, 2020-07-20:
Fixed by a853fff838c1f995b9eeb0ebef5f117cfa3d5f0a

PR #6249 (2020-07-20): py/obj.h: Fix mp_seq_replace_slice_no_grow to use memmove not memcpy.
Fixes #6244
closed 2020-07-20 by dpgeorge