micropython/micropython #6245: gc_alloc_amount  Variable not subtract by gc_free

zxc199108, 2020-07-15:
I find gc_alloc_amount just clear by gc_collect but，not subtract when gc_free. so the gc_alloc_amount is not accuracy. I suggest when MICROPY_GC_ALLOC_THRESHOLD == 1 ,we can compute gc_alloc_amount at gc_free.

dpgeorge, 2020-07-16:
> I find gc_alloc_amount just clear by gc_collect but not subtract when gc_free. so the gc_alloc_amount is not accuracy

`gc_alloc_amount` is not supposed to be a count of the total allocated memory, but rather the amount of memory that has been advanced over since the last GC collection.  The idea is that when this passes the threshold it means that there may be lots of memory to collect, even if there are some holes from calls to `gc_free`.

zxc199108, 2020-07-17:
> `gc_alloc_amount` is not supposed to be a count of the total allocated memory, but rather the amount of memory that has been advanced over since the last GC collection. The idea is that when this [...]
 
I use **gc.threshold(1024)**，because parse.c use gc_free, so gc_alloc_amount increase very fast，and auto gc trigger.

print some error:
*** assertion failed: ATB_GET_KIND(block) == AT_HEAD   \py\gc.c, line 593

dpgeorge, 2020-07-19:
> print some error:
> *** assertion failed: ATB_GET_KIND(block) == AT_HEAD \py\gc.c, line 593

What port/hardware do you use when you get this error?  Do you have a way to reproduce this error that you can share here?

zxc199108, 2020-07-20:
> > print some error:
> > *** assertion failed: ATB_GET_KIND(block) == AT_HEAD \py\gc.c, line 593
> 
> What port/hardware do you use when you get this error? Do you have a way to reproduce this error that you can share here?

I porting code to stm32l476-nucleo，my OS is RT-thread. I alloc 5K ram for MicroPython test auto GC.

this is my code:
```
import gc
gc.threshold(1024)
print('test gc')

GC: total: 4992, used: 288, free: 4704
 No. of 1-blocks: 8, 2-blocks: 5, max blk sz: 2, max f    ree sz: 284
GC: total: 4992, used: 160, free: 4832
 No. of 1-blocks: 4, 2-blocks: 3, max blk sz: 2, max f    ree sz: 298

GC: total: 4992, used: 1232, free: 3760
 No. of 1-blocks: 5, 2-blocks: 6, max blk sz: 32, max     free sz: 234
GC: total: 4992, used: 592, free: 4400
 No. of 1-blocks: 5, 2-blocks: 6, max blk sz: 8, max f    ree sz: 234
*** assertion failed: i == n, \kernel\py\parse.c, line 954

SIGABRT: Abnormal termination

```

dpgeorge, 2020-07-20:
Since you're making a custom port with an RTOS then it's most likely that the problem lies with the way that you integrated MicroPython.  To make progress you'll need to debug the issue and find the root cause.  Try increasing MicroPython's heap to 16k and run the MicroPython test-suite.  Also make sure there is enough stack, around 4k, for the MicroPython task/thread.

zxc199108, 2020-07-20:
> Since you're making a custom port with an RTOS then it's most likely that the problem lies with the way that you integrated MicroPython. To make progress you'll need to debug the issue and find the [...]

I had find my error in **gccollect.c**, **rt_thread_self()->stack_addr** is a const addr, **rt_thread_self()->sp** is 
dynamic. Maybe my OS thread->sp have some error. Thanks your reply. 
```
//PASS CODE
gc_collect_root(rt_thread_self()->stack_addr, ((mp_uint_t)((void *)MP_STATE_THREAD(stack_top) - rt_thread_self()->stack_addr)) / 4);

//NO GOOD CODE
gc_collect_root(rt_thread_self()->sp, ((mp_uint_t)((void *)MP_STATE_THREAD(stack_top) - rt_thread_self()->sp)) / 4);
```

closed 2020-07-20 by dpgeorge