micropython/micropython #6255: Problem while building mpy-cross for ESP32.

tionebrr, 2020-07-18:
Hello folks,
I can't build mpy-cross for ESP32 (IDF 4c81978a3e2220674a432a588292a4c860eef27b) from a container.

Here is the sauce.
```
root@8a77d42bf3ca:~/micropython/mpy-cross# make V=1       
python3 ../py/makeversionhdr.py build/genhdr/mpversion.h
GEN build/genhdr/qstr.i.last
/root/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -E -I. -Ibuild -I.. -Wall -Werror -Wpointer-arith -Wuninitialized -std=gnu99 -DFFCONF_H=\"lib/oofatfs/ffconf.h\" -Os   -fdata-sections -ffunction-sections -fno-asynchronous-unwind-tables -DNO_QSTR -Ibuild/tmp ../py/mpstate.c ../py/malloc.c ../py/gc.c ../py/pystack.c ../py/qstr.c ../py/vstr.c ../py/mpprint.c ../py/unicode.c ../py/mpz.c ../py/reader.c ../py/lexer.c ../py/parse.c ../py/scope.c ../py/compile.c ../py/emitcommon.c ../py/emitbc.c ../py/asmbase.c ../py/asmx64.c ../py/emitnx64.c ../py/asmx86.c ../py/emitnx86.c ../py/asmthumb.c ../py/emitnthumb.c ../py/emitinlinethumb.c ../py/asmarm.c ../py/emitnarm.c ../py/asmxtensa.c ../py/emitnxtensa.c ../py/emitinlinextensa.c ../py/emitnxtensawin.c ../py/formatfloat.c ../py/parsenumbase.c ../py/parsenum.c ../py/emitglue.c ../py/persistentcode.c ../py/runtime.c ../py/runtime_utils.c ../py/scheduler.c ../py/nativeglue.c ../py/pairheap.c ../py/ringbuf.c ../py/stackctrl.c ../py/argcheck.c ../py/warning.c ../py/profile.c ../py/map.c ../py/obj.c ../py/objarray.c ../py/objattrtuple.c ../py/objbool.c ../py/objboundmeth.c ../py/objcell.c ../py/objclosure.c ../py/objcomplex.c ../py/objdeque.c ../py/objdict.c ../py/objenumerate.c ../py/objexcept.c ../py/objfilter.c ../py/objfloat.c ../py/objfun.c ../py/objgenerator.c ../py/objgetitemiter.c ../py/objint.c ../py/objint_longlong.c ../py/objint_mpz.c ../py/objlist.c ../py/objmap.c ../py/objmodule.c ../py/objobject.c ../py/objpolyiter.c ../py/objproperty.c ../py/objnone.c ../py/objnamedtuple.c ../py/objrange.c ../py/objreversed.c ../py/objset.c ../py/objsingleton.c ../py/objslice.c ../py/objstr.c ../py/objstrunicode.c ../py/objstringio.c ../py/objtuple.c ../py/objtype.c ../py/objzip.c ../py/opmethods.c ../py/sequence.c ../py/stream.c ../py/binary.c ../py/builtinimport.c ../py/builtinevex.c ../py/builtinhelp.c ../py/modarray.c ../py/modbuiltins.c ../py/modcollections.c ../py/modgc.c ../py/modio.c ../py/modmath.c ../py/modcmath.c ../py/modmicropython.c ../py/modstruct.c ../py/modsys.c ../py/moduerrno.c ../py/modthread.c ../py/vm.c ../py/bc.c ../py/showbc.c ../py/repl.c ../py/smallint.c ../py/frozenmod.c ../extmod/moduasyncio.c ../extmod/moductypes.c ../extmod/modujson.c ../extmod/modure.c ../extmod/moduzlib.c ../extmod/moduheapq.c ../extmod/modutimeq.c ../extmod/moduhashlib.c ../extmod/moducryptolib.c ../extmod/modubinascii.c ../extmod/virtpin.c ../extmod/machine_mem.c ../extmod/machine_pinbase.c ../extmod/machine_signal.c ../extmod/machine_pulse.c ../extmod/machine_i2c.c ../extmod/machine_spi.c ../extmod/modbluetooth.c ../extmod/modussl_axtls.c ../extmod/modussl_mbedtls.c ../extmod/modurandom.c ../extmod/moduselect.c ../extmod/moduwebsocket.c ../extmod/modwebrepl.c ../extmod/modframebuf.c ../extmod/vfs.c ../extmod/vfs_blockdev.c ../extmod/vfs_reader.c ../extmod/vfs_posix.c ../extmod/vfs_posix_file.c ../extmod/vfs_fat.c ../extmod/vfs_fat_diskio.c ../extmod/vfs_fat_file.c ../extmod/vfs_lfs.c ../extmod/utime_mphal.c ../extmod/uos_dupterm.c ../lib/embed/abort_.c ../lib/utils/printf.c build/genhdr/moduledefs.h ../py/mpconfig.h mpconfigport.h >build/genhdr/qstr.i.last
../py/persistentcode.c:845:2: error: #error mp_raw_code_save_file not implemented for this platform
 #error mp_raw_code_save_file not implemented for this platform
  ^~~~~
make: *** [../py/mkrules.mk:89: build/genhdr/qstr.i.last] Error 1
make: *** Deleting file 'build/genhdr/qstr.i.last'
```

Maybe an easy fix? Any clue is welcome really.

jimmo, 2020-07-18:
mpy-cross should be built for your host platform, not the target. It looks like you're compiling with the cross-compiler, rather than regular gcc.

tionebrr, 2020-07-19:
Hey thanks @jimmo !
Solved :)

closed 2020-07-19 by tionebrr