micropython/micropython #6257: STM32H7 SPI6 not working with DMA

rcordeau, 2020-07-19:
Hi,
I was trying this out on a custom board as well as the Nucleo STM32H743SZI board, and while SPI3 works when sending multiple bytes (default behavior when sending more than 1 byte seems to be to use DMA per spi.c), when it comes to SPI6, it uses a different DMA engine (DMA2) and goes through a mux BDMA_REQUEST_SPI6_TX/RX. I keep getting a ETIMEDOUT error due to waiting for the interrupt to come back from the DMA transaction. Any advice, or could this be a bug?

Thanks,
Rob

rcordeau, 2020-07-20:
I think it has something to do with the fact that SPI6 DMA uses a BDMA channel instead of a DMA stream like the other peripherals.

rcordeau, 2020-11-11:
Anybody?

rcordeau, 2022-01-28:
Hello?

pankaj-xscape, 2024-05-28:
Facing the same issue. SPI6 works standalone, but not with BDMA.

boringethan, 2025-01-08:
Same thing here. A call to the HAL_SPI_Receive_DMA triggers the HAL_SPI_ErrorCallback and HAL_SPI_ERROR_DMA with a HAL_DMA_ERROR_TE are raised. What gives?