micropython/micropython #6261: Why MicroPython isn't tailored from CPython

yang8621, 2020-07-20:
Hi,

Just noticed MicroPython, which is very helpful for embedded systems. 
Just for my curiosity, why create micropython from zero and then to keep compatibility with CPython? 

Why not tailor CPython to "MicroPython" as a minimal set? 

Thanks.

stinos, 2020-07-20:
The way I see it: CPython is tied to the standard, getting it to adopt a minimal set somehow would probably simply be rejected - because it's non-standard - and if not would take years to get accepted. Moreover large parts of the CPYthon source code are just not suitable for running on embedded systems. The effort it would take to change all of that could easily take longer than just starting from scratch.

Note: questions like these aren't issues, merely questions, and as sucg the forum is better suited to ask: https://forum.micropython.org

jimmo, 2020-07-20:
If you're interested in the history of MicroPython, this talk is very good -- https://youtu.be/Zm08hXeuv-I

MicroPython is not simply a "cut down CPython", it's a very different approach to the implementation by choosing different trade-offs. If you have questions you're welcome to ask on the forum.

closed 2020-07-20 by jimmo

yang8621, 2020-07-21:
Thank you. I will ask in the forum.