micropython/micropython #6266: Socket.sendto() never transmitting in on ESP32

j4mag, 2020-07-21:
Hello all,

I'm writing some code in MicroPython for an ESP32-Devkitc-S1 that will need to send and receive UDP packets.

I've tested the code in Jupyter notebooks, and it's capable of both sending and receiving packets with my test socket. 

On the ESP32, the same code is able to receive UDP packets just fine, but is incapable of sending packets (I've tried a few dozen times, so it doesn't seem like they're just dropped). The code runs without errors, and sock.sendto returns 17 as I'd expect, but the packets are **never** received. 

I have an access point quite close to me, and the board is able to connect to the wifi without much hassle, so I don't think its an issue with the transmissions being too weak.

Please let me know if you have any suggestions, or if this is a known issue.

`
import network
import socket
import time

#I only run this on the SoC since my laptop is already connected to the wifi.
wlan = network.WLAN(network.STA_IF)
wlan.active(True)
wlan.ifconfig(("192.168.100.102","255.255.255.0","192.168.1.1","8.8.8.8"))
wlan.connect("ap_ssid","ap_pass") #with my actual wifi credentials

def recv_packet(): #works on desktop and on SoC
    sock = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
    sock.bind(("",7777))
    print("Waiting...")
    data, addr = sock.recvfrom(2048)
    print("Received: ", data, "from", addr)
    sock.close()
    time.sleep(3) #the socket needs time to close or we run out of mem

def send_packet(): #works on desktop, not on SoC
    sock = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
    sock.setsockopt(socket.SOL_SOCKET, socket.SO_REUSEADDR, 1)
    data = b'testing in Python'
    addr = ("192.168.39.136", 7778) #a UDP socket on my phone to test
    sock.sendto(data, addr)
    print("Sent: ", data, "to", addr)
    sock.close()
    time.sleep(3) #the socket needs time to close or we run out of mem
`

peterhinch, 2020-07-22:
Adapted for my network this works with a Pyboard D as the transmitter and my laptop as receiver, so I guess:
 * There's nothing wrong with the code.
 * The problem is ESP32 specific.

j4mag, 2020-07-22:
Thanks for checking that out Peter, I appreciate it. 
It sounds like it is an ESP32 issue. 

I'm running the esp32-idf3-20190125-v1.10 release of the MicroPython binaries (for the same reasons described in [this thread](https://github.com/micropython/micropython/issues/5336)), and it's possible this isn't an issue in v1.12, but I can't be sure.

Two-way communication with the board would be useful but not necessary for my project, so I'll roll with the punches, but I'm still interested in figuring this out.