micropython/micropython #6267: py/compile: __aiter__ special method shouldn't be async

jonathanhogg, 2020-07-21:
So the `__aiter__()` method should return an object that supports the `__anext__()` method. According to the [Python documentation](https://docs.python.org/3.5/reference/datamodel.html#object.__aiter__), the latter of these must either return an awaitable or be a coroutine (`async def`), but the former is just a regular method.

With this simple example (coded to work in either CPython or MicroPython):

```python
try:
    import asyncio
except ImportError:
    import uasyncio as asyncio

class SlowCounter:
    def __init__(self, n):
        self.n = n

    def __aiter__(self):
        self.i = 0
        return self

    async def __anext__(self):
        await asyncio.sleep(1)
        if self.i == self.n:
            raise StopAsyncIteration
        i = self.i
        self.i += 1
        return i

async def main():
    async for i in SlowCounter(10):
        print(i)

asyncio.run(main())
```

I get the expected behaviour under CPython, but this error on MicroPython:

```
Traceback (most recent call last):
  File "<stdin>", line 29, in <module>
  File "uasyncio/core.py", line 1, in run
  File "uasyncio/core.py", line 1, in run_until_complete
  File "uasyncio/core.py", line 1, in run_until_complete
  File "<stdin>", line 26, in main
TypeError: 'SlowCounter' object isn't iterable
```

Following [this example](https://github.com/peterhinch/micropython-async/blob/master/v3/docs/TUTORIAL.md#42-asynchronous-iterators) from @peterhinch's V3 tutorial I found that the fix is to make `__aiter__()` a coroutine, i.e., just make it an `async def`. However, this seems to be a clear break in behaviour between MicroPython and CPython.

I suspect the problem is this line in the `compile_async_for_stmt()` function:

```python
    compile_await_object_method(comp, MP_QSTR___aiter__);
```

I'm not familiar with the compiler, but I had a guess that it ought to be just calling the method without awaiting the result and tried changing it to:

```python
    EMIT_ARG(load_method, MP_QSTR___aiter__, false);
    EMIT_ARG(call_method, 0, 0, 0);
```

and this works.

I'm not going to put in a PR just yet since this might be subtly wrong and it also has knock-on effects to any code already written that is expecting `__aiter()__` to have to return an awaitable / be a co-routine.

dpgeorge, 2020-07-22:
Thanks for the detailed issue.  I can't recall the exact history/reasoning around the implementation of `__aiter__`, but it was done a while back, around 4.5 years ago, see 81ebba7e0236163b7594938201bf3a6b802ebfaa

Doing some investigation now, it seems that MicroPython's implementation of `__aiter__` was correct for an earlier (provisional) version of PEP492 (CPython 3.5) but that changed in the final version of PEP492 (in CPython 3.5.2). See https://www.python.org/dev/peps/pep-0492/#why-aiter-does-not-return-an-awaitable .  See also the note at the end of this subsection in the docs: https://docs.python.org/3.5/reference/datamodel.html#asynchronous-iterators .  And for completeness: https://bugs.python.org/issue27243

To be consistent with the Python spec as it stands today (and now that PEP492 is final) MicroPython's behaviour should be changed to match.

jonathanhogg, 2020-07-22:
Interesting! I hadn't seen the storied history of `__anext__()`. I was still doing all my (commercial) coding in Python 2 around this time and eyeing Python 3 enviously from afar.

Should this have a similar automatic conversion and `DeprecationWarning` or is the async stuff considered in-flux enough in MicroPython that it's OK to just change?

If the latter, and my fix looks OK to you, then I'm happy to submit a PR for it. I'd also be equally happy to leave it to a core developer with more expertise in messing with the compiler 😉

jonathanhogg, 2020-07-22:
I've opened a PR anyway. @dpgeorge: feel free to take or leave as you wish. If this is going to happen later as part of a more considered update then it'd be good to know a timeframe so I can figure out what to do with my current code that uses `__aiter__()`.

@peterhinch: I'm happy to open a PR on your V3 tutorial as well, but I don't know if you want to do that yourself or hang on for this (or equivalent) change to hit a production release.

On a related note: is there a plan to implement asynchronous generators any time soon? I know it's pretty esoteric stuff for most people, but I've already written three different async-iterable classes in the last week and each time I thought, "this would be so much simpler with an async generator…". Async code is just such a natural fit for microcontrollers and trying to implement any serious logic using the BLE irq callback is a massive chore without it. If nobody's looking at it, I'd be happy to make it a side-project and an excuse to stare harder at how the MicroPython compiler and runtime works.

peterhinch, 2020-07-22:
@jonathanhogg My V3 tutorial aims to match the current daily build. I'll update it when your PR is accepted.

dpgeorge, 2020-07-24:
Fixed by 37e1b5c891f9964bb6c95228bc2d718511507a69

closed 2020-07-24 by dpgeorge

dpgeorge, 2020-07-25:
> On a related note: is there a plan to implement asynchronous generators any time soon?

No, there is currently no plan for this.  The main issue to overcome here is that currently MicroPython mixes the concepts of generators and coroutines, ie `await` is just `yield from` and there's no explicit concept of an async function - as far as the runtime is concerned it's just a generator.  So for an async generator there'd need to be a way to distinguish between `yield` meaning yield a value, and `yield from` meaning `await`.

jonathanhogg, 2020-07-25:
> No, there is currently no plan for this. The main issue to overcome here is that currently MicroPython mixes the concepts of generators and coroutines, ie `await` is just `yield from` and there's no [...]

Yeah, I started to have a look at this and tried to get my head around the spec in PEP 525 and the CPython design, but realistically I've not got the time to devote to this at the moment. I'm keeping it in the back of my mind, though, in case I get bored. I guess creating a co-routine type in MicroPython and separating out `await` and `yield from` would be the place to start.