micropython/micropython #6271: Prepare for next release v1.13

dpgeorge, 2020-07-22:
It's been a long time (too long... but this year has been very different...) since the last release v1.12 on 20th Dec 2019.  So time for the next one which would be v1.13.  There are a lot of changes since v1.12 but nothing in particular that would warrant a move to v2.0.

After v1.13 the aim would be for more frequent releases, ideally only 1-2 months between each one.

The main outstanding items to address before v1.13 are:
- [ ] #6061 fix GC issue with some native .mpy files
- [ ] #6114 timestamps on littlefs filesystems (possibly, not sure about this)

dpgeorge, 2020-07-22:
See #4821 for previous release discussion.

peterhinch, 2020-07-22:
Timestamps would be very useful, particularly for users of rshell's **rsync** command.

Walkline80, 2020-07-22:
Is esp32 ble pairing & bonding added to the schedule?

kevinkk525, 2020-07-24:
all uasyncio related issues and PRs

jacklinquan, 2020-07-28:
Really looking forward to this stable version that supports BLE for ESP32. So projects can go more mature and be based on a stable code base.

rcordeau, 2020-07-29:
Any chance for a bit more STM32H7 support? Perhaps pulling in that ETH fix and maybe checking out the BDMA SPI6 issue?  Pretty please?

xzjkya, 2020-08-28:
Is there any release date intent/prevision for v1.13? Looking for a more "stable" uasyncio release.

jcw, 2020-09-03:
Bravo and congrats for getting [v1.13](https://forum.micropython.org/viewtopic.php?f=8&p=50627#p50627) done, Damien! \o/

dpgeorge, 2020-09-03:
Thanks for everyone's patience, and for everyone who was involved in contributing in some way to the code.  v1.13 was tagged in b0932fcf2e2f9a81abf7737ed4b2573bd9ad4a49 

The plan moving forward is to have a more strict, 2 month release cycle with a focus on specific features for the release.

closed 2020-09-03 by dpgeorge