micropython/micropython #6281: STM32F765 support

vzasukhin, 2020-07-24:
I would like to add support for STM32F765. Linker script and alternative functions for STM32F765 are compatible with STM32F767 (which is already supported in `ports/stm32/boards/stm32f767{.ld,_af.csv}`).
How should I add support for STM32F765: copy and rename files or create symbolic links to stm32f767{.ld,_af.csv}?

dpgeorge, 2020-07-24:
> How should I add support for STM32F765: copy and rename files or create symbolic links to stm32f767{.ld,_af.csv}?

If the files are equivalent for the F765 (ie the F765 can work by using the F767 files) then just use the F767 files directly.  Some other MCUs do this, eg the F446 uses the F429 files.

rolandvs, 2020-07-24:
It is as damien indicates. However if you have some different memory map it might be handy to make the ld files local to the board. I use the internal Flash for a project differently than the common ld files on several processors so I like the idea to move them to the specific board directory as well as the manifest related files.

vzasukhin, 2020-07-31:
It would be convenient if compatible chips were indicated somewhere. Comparing existing files with datasheets is time consuming. Is it worth creating README.rst in ports/stm32/boards and keep files compatibility there?