micropython/micropython #6295: writing is not permitted when use ble_simple_peripheral demo

pzg250, 2020-07-29:
Hi,
I am trying to use  ble_simple_peripheral.py. But when I use mobile to connect to esp32. It can connect to esp32. But cannot write value. The tool shows "writing is not permitted"Any advise ?  ( I use nRF Connect.)

Regards,
Major

[image]

[image]

jimmo, 2020-07-29:
You should be writing to the characteristic with UUID "6E400002-B5A3-F393-E0A9-E50E24DCCA9E" -- doesn't look like you're looking at the right service in nRF Connect.

pzg250, 2020-07-29:
> You should be writing to the characteristic with UUID "6E400002-B5A3-F393-E0A9-E50E24DCCA9E" -- doesn't look like you're looking at the right service in nRF Connect.

Hi, thank you. 
Yes, but only UUID 2902 available.

pzg250, 2020-07-29:
Hi @jimmo , could you indicate how to do next ? Maybe need to change the demo code ? or use some some special mobile app?

pzg250, 2020-08-08:
Anyone can help? Or is there a doable demo ? Thanks in advance.

pzg250, 2020-08-21:
use latest master version.

closed 2020-08-21 by pzg250