micropython/micropython #6296: natmod: Document pyelftools dependency

jimmo, 2020-07-29:
mpy_ld.py needs pyelftools v0.26

Suggestion to do is have the mpy_ld.py detect:
 - Whether it can `import elftools`
 - What version is available

and warn the user appropriately (rather than failing when accessing the `data_size` field on `Section` -- see https://github.com/micropython/micropython/issues/6264#issuecomment-662288371 where this fails mysteriously on Ubuntu 18.04 as it has v0.24).

stinos, 2020-07-29:
Could also write this in the README where the prerequisites are and/or where the natmod docs are. At least those are the places I went looking for info when I had the problem.

remixer-dec, 2020-09-19:
Yeah, based on the error, I thought that I needed elftools module, but elftools from pypi is completely unrelated.
Since there is no information about requirements for native module build in the docs, I'll describe what I had to do, to build the [sample code](https://docs.micropython.org/en/latest/develop/natmod.html) from the docs, if someone from non-native world will be searching for this.  
  
 *Building native modules for ESP32 on Windows*

1) get folders *py* and *tools* from the root of micropython repo  
2) make sure that paths in *makefile* are correct and edit *makefike*, so ```MPY_DIR``` leads correctly to the main directory with above-mentioned micropython folders
3) copy *py* folder to your project location, or edit the include line (from the sample)  
4) get [cygwin](https://cygwin.com/), install *make*, *python3* for it, run cygwin  
5) run ```python3 -m ensurepip``` to install pip (and avoid conflicts with windows-based pip)  
6) run ```python3 -m pip install pyelftools```  
7) get [ESP32 toolchain](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/windows-setup-scratch.html#toolchain-setup)  
8) unzip it somewhere in cygwin dir, for example in ~ (/home/username)  
9) add toolchain path to cygwin PATH env. variable ```export PATH=/home/username/xtensa-esp32-elf/bin/:$PATH```  
If you want to make it permanent, add that to *~/.profile* file  
10) run ```make ARCH=xtensawin```  
  
P.S. I had some cygwin modules already installed, so maybe I missed something.  
P.S.S. There are probably other ways to use make on Windows, like WSL/MingW instead of cygwin

dpgeorge, 2020-09-23:
See #6475 for an addition to the docs.

dpgeorge, 2020-09-25:
Docs updated in c8ade2bd7f5d0e089098bcc07eb77a770f3da726

closed 2020-09-25 by dpgeorge