micropython/micropython #6297: BLE.active() change gap_name to "ESP32"

jpliew, 2020-07-29:
May I know if it is the natural behavior of BLE.active() to change gap_name please? After BLE.active() the gap_name will always changed back to "ESP32"

Tested on the latest unstable and stable version.

Thanks

```
>>> import os
>>> os.uname()
(sysname='esp32', nodename='esp32', release='1.12.0', version='v1.12-662-g8da40baa4 on 2020-07-29', machine='ESP32 module (spiram) with ESP32')
>>> import bluetooth
>>> ble=bluetooth.BLE()
>>> ble.config(gap_name="TESTING")
>>> ble.config('gap_name')
b'TESTING'
>>> ble.active(True)
I (2350) BTDM_INIT: BT controller compile version [c1cbe45]
I (6316) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
W (6326) phy_init: failed to load RF calibration data (0x1102), falling back to full calibration
I (6476) phy: phy_version: 4180, cb3948e, Sep 12 2019, 16:39:13, 0, 2
True
>>> ble.config('gap_name')
b'ESP32'
>>>
```

jha929, 2020-10-27:
Hi, I faced the same issue and have looked at the low-level code of BLE.active method. It's simply setting the gap_name to their default, "ESP32" in my case.

However, according to the [documentation](https://docs.micropython.org/en/latest/library/ubluetooth.html#ubluetooth.BLE.gatts_read), the BLE's gap_name can be set anytime.

Therefore, calling the BLE.active method first followed by BLE.config(gap_name="YOUR_GAP_NAME") will do the trick.

An example code snippet is given below.

```commandline
>>> ...
>>> ble.active(True)
>>> ble.config('gap_name')
b'ESP32'
>>> ble.config(gap_name="TESTING")
>>> ble.config('gap_name')
b'TESTING'
```

jpliew, 2020-10-27:
@jha929 yes I did the same to fix it.

Not sure how the micropython team wants this to be fixed, by adding a check for gap_name in BLE.active or just put a document saying that the gap_name should be set after BLE.active.

mohnazemi, 2022-03-07:
I faced the same issue recently. The latest [documentation](https://docs.micropython.org/en/latest/library/bluetooth.html) still says:

> 'gap_name': Get/set the GAP device name used by service 0x1800, characteristic 0x2a00. This can be set at any time and changed multiple times.

The GAP name changes only takes effect after calling active(True) on BLE

saepien, 2023-01-31:
Does not work any which way. Name still comes up as ESP32 on BLE scanner apps.

saepien, 2023-01-31:
Only worked after I changed the name to ALL UPPERCASE LETTERS!@@

silbo, 2023-10-17:
I have the issue that on Mac OS always ESP32 is shown as the name. On windows and linux it works to change it. May be I am doing it wrong.

mohnazemi, 2023-10-18:
@silbo Probably this is due to caching on iOS/Mac Bluetooth stack.
There are two names to consider for a peripheral: the advertising name and the GAP name. For a peripheral you never connected to, the 'name' will be the advertising name. Once it is connected, the GAP name is cached and it is instead reported as the peripheral's name. Changing the GAP name after that may not show up on a central until that cache is invalidated.

closed 2023-12-22 by dpgeorge