micropython/micropython #6298: Deepsleep() on nrf52832 not implemented?

Deenis, 2020-07-29:
I'm having a custom build hardware with the nrf52832, everything works so far, except deepsleep.

Pulling latest from repository, and building with:
`make BOARD=feather52 FROZEN_MPY_DIR=freeze SD=s132 flash sd`

Connecting to device through WebBluetooth REPL, everything works as expected. However:
`import machine`
`machine.deepsleep(60000)`

Throws the following error:
`Traceback (most recent call last):`
`  File "<stdin>", in <module>`
`TypeError: function takes 0 positional arguments but 1 were given`

And the following has no effect:
`import machine`
`machine.deepsleep()`

Same goes with:
`import machine`
`machine.lightsleep()`

Are these features not implemented yet?

glennrub, 2020-07-30:
> import machine
machine.deepsleep(60000)

>Throws the following error:
Traceback (most recent call last):
File "<stdin>", in <module>
TypeError: function takes 0 positional arguments but 1 were given

This is because the `deepsleep()` does not support a parameter. 

> And the following has no effect:
import machine
machine.deepsleep()

In the nrf port `deepsleep()` is an alias for the `__WFI`. And `lightsleep()` an alias for `__WFE`. I can only speculate at this point, but i guess the reason why it does not go to any sleep mode could be events from SoftDevice during echoing the characters back to the WebBluetooth REPL causing it to wake up (`BLE_EVT_TX_COMPLETE` event). Testing `deepsleep()` on UART REPL it seems to work as expected. It wakes up on the next key press.

I believe you should still be able to use `deepsleep()`  with Bluetooth LE if you freeze the application instead of issuing the commands over the interactive WebBluetooth REPL.

Deenis, 2020-07-31:
That seems to be the cause of my problems - thank you :)

closed 2020-07-31 by Deenis