micropython/micropython #6303: framebuf.MONO_VMSB missing

emard, 2020-08-03:
framebuf.MONO_VLSB exists but
framebuf.MONO_VMSB is missing but needed for e-paper 2.9" display

peterhinch, 2020-08-03:
[This test script](https://github.com/mcauser/micropython-waveshare-epaper/blob/master/examples/2in9-hello-world/test.py) for @mcauser's driver uses **framebuf.MONO_HLSB**.

emard, 2020-08-03:
Yeah great but when hardware rotation of the screen different than this
driver (initialized screen in Y-mode, this driver initializes in X-mode)

There has been long discussin on here
https://forum.micropython.org/viewtopic.php?t=6319

All this mess and manual rotation of the screen would
not be neccessare if micropython had MONO_VMSB

MicroPython v1.12 on 2019-12-20; ESP32 module with ESP32
>>> import framebuf
>>> framebuf.MONO_VLSB
0
>>> framebuf.MONO_VMSB
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'module' object has no attribute 'MONO_VMSB'

peterhinch, 2020-08-03:
OK, fair point.

Dreagonmon, 2020-09-05:
https://github.com/Dreagonmon/micropython/tree/patch_framebuf_mvmsb