micropython/micropython #6311: Support for ESP-Mesh

seisatsu, 2020-08-05:
ESP-Mesh allows support for mesh networking between ESP devices. The last mention I can find of mesh networking as a possibility for MicroPython is https://github.com/micropython/micropython-esp32/issues/176 from three years ago, which was closed without being implemented. I'd like to request this feature if it isn't already on the table.

60999, 2020-11-29:
Is there any update progress?

rubydesign, 2021-03-28:
Just guessing that this will NOT drop out of the sky (though i too would like that). It has been closed several times and does require a fair amount of work. 

The good news is (o think) that esp-mesh is just espressifs implementation of a mesh. Just like painlessmesh.
I guess a mesh is not as difficult to build in python as in cpp (so it "just" needs the necessary API to be available)
Another guess is that once the esp-now code is merged, it should be relatively easy.
So let's root for #6515 to be merged and then someone can build another python layer on top of that to make meshes even easier.

just 2c from a newbie