micropython/micropython #6314: Bug in esp32 viper code emitter

tve, 2020-08-06:
The viper code emitter for the esp32 seems to have a bug causing an expression `data[IX] << 8 | data[IX+1]` where `data = ptr8(...)` to produce a bogus value. Test case:
```
data = bytearray([100, 101, 102, 103, 104])

IX = const(3)

def normal_ix():
    a = data[IX]
    b = data[IX+1]
    c = data[IX] << 8 | data[IX+1]
    return a,b,c

@micropython.viper
def viper_ix():
    d = ptr8(data)
    a = int(d[IX])
    b = int(d[IX+1])
    c = d[IX] << 8 | d[IX+1] # <-- BUGGY
    h = d[IX] << 8
    i = d[IX+1]
    j = h|i # <-- correct value for c
    return a,b,c, h,i,j

print(normal_ix())
print(viper_ix())
```
Produces:
```
> pyboard bug.py
(103, 104, 26472)
(103, 104, 1065454444, 26368, 104, 26472)
```
The second line should have been: `(103, 104, 26472, 26368, 104, 26472)`

I couldn't figure out how to get a disassembly of the code produced so didn't look into what gets generated.

tve, 2020-08-06:
More problems:
```
data = bytearray([100, 101, 102, 103, 104])

IX = const(3)

@micropython.viper
def viper2_ix() -> int:
    d = ptr8(data)
    a = int(d[IX])
    b = int(d[IX+1])
    c = d[IX] << 8
    c |= d[IX+1]
    return c

@micropython.viper
def viper3_ix() -> int:
    d = ptr8(data)
    c = d[IX] << 8
    c |= d[IX+1]
    return c

print(viper2_ix())
print(viper3_ix())
```
Produces:
```
104
26472
```
Expected two times 26472.

peterhinch, 2020-08-06:
I ran these on a Pyboard D. The first script produces the output you expect, confirming the ESP32 bug:
```
(103, 104, 26472)
(103, 104, 26472, 26368, 104, 26472)
```
The second script produces this, suggesting a platform-agnostic bug:
```
104
26472
```
I agree it should produce 26472 twice.

dpgeorge, 2021-09-13:
See #7783 for a fix to the second (port agnostic) issue here.

dpgeorge, 2021-09-14:
The second issue was fixed in 426785a19eeb12aef7383fbda4693575d8c4dddf

jimmo, 2022-11-09:
This was just reported on the Discord. Here's a simple repro

```
@micropython.viper
def foo(bptr: ptr8) -> int:
  return (bptr[0] << 8) + bptr[1]

print(foo(b'\x01\x01'))
```

Prints `1073630898` on esp32. Correctly prints 257 on stm32/unix.

dpgeorge, 2022-11-11:
Here's a version of the above that also fails on unix x86-64 (but not stm32):
```py
@micropython.viper
def f3(b: ptr8) -> int:
    return b[0] << 24 | b[1] << 16 | b[2] << 8 | b[3]

print(hex(f3(b"\x01\x02\x03\x04")))
```

dpgeorge, 2022-11-11:
Fixed by 0698dd72ea3b9d7897e63c6e219061bff8d162cf

closed 2022-11-11 by dpgeorge