micropython/micropython #6315: ESP32 - BT Scanning Broken ?

diginfo, 2020-08-06:
https://forum.micropython.org/viewtopic.php?f=18&t=8842

When I revert back to April build, BT scanning works, with current release nothing is detected.

jimmo, 2020-08-06:
Looks like you're using unstable builds, but your code is still using the v1.12 API. The API has changed (this will be in the v1.13 release notes). In particular your issue is likely the IRQ event numbers have changed. The docs have been updated though, please see:
https://docs.micropython.org/en/latest/library/ubluetooth.html#ubluetooth.BLE.irq

closed 2020-08-06 by jimmo

diginfo, 2020-08-06:
Thanks very much for helping, but changing:

` _IRQ_SCAN_RESULT   = const(1 << 4)`

to

`_IRQ_SCAN_RESULT   = const(6)`

Did not solve the issue and it's still unable to scan

diginfo, 2020-08-06:
OOps - yes that solved it, thanks :-)