micropython/micropython #6316: [NUCLEO_WB55 - USBDONGLE_WB55] disconnect usb

dmazzella, 2020-08-06:
Hi all,
I have tried bluetooth demo from examples:
```
ble_simple_peripheral.py
ble_temperature.py
ble_uart_repl.py
```

on my devices (NUCLEO_WB55 and USBDONGLE_WB55) i have latest version og MP: 
`MicroPython v1.12-663-g9883d8e81 on 2020-08-05; NUCLEO-WB55 with STM32WB55RGV6 `

BLE stack firmware i have tried (V1.8.0) from [STM32CubeWB](https://github.com/STMicroelectronics/STM32CubeWB)
```
stm32wb5x_BLE_Stack_full_fw.bin
stm32wb5x_BLE_HCILayer_fw.bin
```

and in all tests usb devices was disconnected from pc, BLE example works correctly, i can connect to device and in case of `ble_uart_repl.py` repl work correctly.

is this the expected behavior?

Thanks,
D.

dmazzella, 2020-08-28:
A similar issue was reported and solved in zephyr [#25880](https://github.com/zephyrproject-rtos/zephyr/issues/25880)

jimmo, 2020-08-28:
Hi @dmazzella -- I had noticed this also, any WS firmware newer than 1.5+ seems to see this, and didn't see the issue if I was connecting via the ST-Link rather than the USB VCP.

I will investigate trying to adapt the fix from Zephyr.

dmazzella, 2020-08-31:
@jimmo thanks i have applied your path and it worked for me.

dmazzella, 2020-08-31:
@jimmo 
the release of HSEM is not needed?
`LL_HSEM_ReleaseLock(HSEM, CLK48_SEMID, 0);`

jimmo, 2020-08-31:
We never deinitialise the USB peripheral.

closed 2020-09-09 by dmazzella