micropython/micropython #6323: ESP32； Executed error with correct codes

FFtust, 2020-08-08:
**micropython version:** 
commit 4d94fae83322dd1b4197fd770fa9d0b7474ce72b
Author: Damien George <damien.p.george@gmail.com>
Date:   Wed Jul 10 16:48:37 2019 +1000
**Hardware**
esp32

I encountered an incredible problem. 
Here is the implemetion of "sent_float" in neurons_protocol/neurons_data_conversion.py
 
[image]

and here is the Traceback 

[12:28:22.141]收←◆Traceback (most recent call last):
  File "main1.py", line 29, in <module>
  File "mbuild_modules/standard_shield.py", line 28, in dc_motor_set_power_both
  File "neurons_protocol/neurons_engine.py", line 355, in neurons_request
  File "neurons_protocol/neurons_engine.py", line 247, in common_neurons_command_request
  File "neurons_protocol/neurons_data_conversion.py", line 248, in request_data_conversion
  File "neurons_protocol/neurons_data_conversion.py", line 65, in send_float
NameError: local variable referenced before assignment

I can't find any problem with my codes, but it did happend from time to time. 

Any idea about it, thanks very mush for your help.

FFtust, 2020-09-07:
Sorry， I have not solved the problem， it still occured in my application. 

And I can also see some other logs,
like this,
**File "events/event_manager.py", line 156, in __event_cb_task 
File "events/event_manager.py", line 156, in __event_cb_task 
File "events/event_manager.py", line 121, in __event_cb_task 
File "run_app.py", line 127, in default_task 
File "<string>", line 1, in <module>
File "main1.py", line 29, in <module> 
File "mbuild_modules/standard_shield.py", line 28, in dc_motor_set_power_both 
File "neurons_protocol/neurons_engine.py", line 353, in neurons_request 
File "neurons_protocol/neurons_engine.py", line 221, in common_neurons_command_request 
AttributeError: 'bytearray' object has no attribute 'append'**

or  like this

**assertion "ATB_GET_KIND(block) == AT_HEAD" failed: file "D:/work/mpy_new/mpy/submodule/micropython/py/gc.c", line 588, function: gc_free
abort() was called at PC 0x401ab01b on core 0**

_backtrace for this crash: backtrace: 40094708 <invoke_abort>:
40094ad4 <abort>:
401aaff4 <__assert_func>:
400d2ddc <gc_free$part$0>:
400d340c <gc_free>:
400d2d18 <m_free>:
400e3628 <fun_bc_call>:
400df4cc <mp_call_function_n_kw>:
400ec174 <mp_execute_bytecode>:
400e3628 <fun_bc_call>:
400df52c <mp_call_method_n_kw>:
400ec174 <mp_execute_bytecode>:
400df500 <mp_call_function_0>:
400dff74 <mp_parse_compile_execute>:
400ea174 <do_load>:
400ea1d8 <mp_builtin___import__>:
400dfe3c <mp_import_name>:
400ec174 <mp_execute_bytecode>:
400ea640 <eval_exec_helper>:
400ea6e4 <mp_builtin_exec>:
400e35cc <fun_builtin_var_call>:
400e1cd4 <mp_call_method_self_n_kw>:
400e1d3c <bound_meth_call>:
400ebf20 <thread_entry>:
400f80a0 <freertos_entry>:
4009a6ac <vPortTaskWrapper>:_

I  also find that if I close some modules (like Bluetooth in esp32), the issue doesn't occure anymore. But in my application, MP and bluetooth are completely independent modules.

Thanks for your kind help.

peterhinch, 2020-09-07:
I suggest you check for free RAM and for RAM fragmentation using the **gc** and **micropython** modules. This sort of error can occur occasionally if RAM is very tight. Periodically calling **gc.collect()** can help with fragmentation. Does your hardware have SPIRAM?

FFtust, 2020-09-07:
@peterhinch Thanks vary mush for your reply.
Yes, my hardware has a 4M SPIRAM, and micropython use 1.5M as it's heap.  
In my application,  gc.collect() is called every 10 seconds. I realized that the gc module may cause this kind of issue, but I don't know how to debug.

peterhinch, 2020-09-08:
>  SPIRAM is known to have issues on esp32.

Yes, myself and @kevinkk525 have experienced random crashes in long running tests. These do not occur running the same code on non-SPIRAM boards, although they do occasionally reboot.

kevinkk525, 2020-09-08:
Yeah I can confirm this "great" error.. The esp32 with SPRAM can work fine for a day and then suddenly it tells you that e.g. a Lock object doesn't have the method "release", although it is being called every 20ms..
It's a typical memory error with spram units. Only workaround I found is to reset the device..

FFtust, 2020-09-09:
@dpgeorge  I am  tesing based on  the version v1.13 now. But unfortunately, it still happend.
@peterhinch @kevinkk525 Thanks for your reminder， I will look into the SPIRAM unit. 

The hardware may cause this problem, I will search helps from esp32 team. Thanks for all of your helps.

closed 2020-09-09 by FFtust

FFtust, 2020-09-10:
Hi, it seems that the esp32 team has resolved the spiram issue by updating the toolchain (https://dl.espressif.com/dl/esp32_win32_msys2_environment_and_esp2020r2_toolchain-20200601.zip), but some errors occurred when I used this toolchain to build micropthon ( The idf example has no errors).
Some logs like this.

[image]

Sorry I am not familiar with makefile, does anyone know what the problem is?It seems the error occured when making .cpp files.

kevinkk525, 2020-09-10:
I was rarely able to build the esp32 firmware myself, therefore I always use the daily builds. Would be great if the SPRAM memory issues would be solved in a more recent toolchain and this is used in daily builds then.

dpgeorge, 2020-09-11:
Yes, would be good to update the IDF to a newer v4.x version, along with an update to the toolchain.

FFtust, 2020-09-15:
@kevinkk525 I used the new version toolchain (https://docs.espressif.com/projects/esp-idf/en/release-v3.3/get-started/windows-setup.html) and everything  is ok, no error occured, no crashing. You can download the toolchain and make a try.

By the way, a new version of esp32 hardware V3 has fixed the PSRAM issue too, and I have tested it, it did solove the PSRAM issue.You can look into https://www.espressif.com/sites/default/files/documentation/ESP32_ECO_V3_User_Guide__EN.pdf

kevinkk525, 2020-09-15:
That sounds interesting, thanks. 

Nice that they fixed their hardware too but there are so many old PSRAM boards in circulation that can only hope that the toolchain fixes the problem well enough.