micropython/micropython #6327: gc chip question

zxc199108, 2020-08-11:
GC: total: 92736, used: 79968, free: 12768
 No. of 1-blocks: 834, 2-blocks: 96, max blk sz: 384, max free sz: 266

the max free sz is 266*16=4256 bytes，but free is 12768 bytes. so when I setup thread(5KB), memory alloc failed.

How can i solve this problem?

peterhinch, 2020-08-11:
You are suffering from RAM fragmentation. It's best to instantiate objects which require a large block of contiguous RAM soon after program start, before fragmentation occurs. That said, you are running very tight on RAM. Have you tried frozen bytecode to reduce RAM usage?

Incidentally queries like this are best raised in [the foum](http://forum.micropython.org/index.php) rather than in this bug tracker.

stinos, 2020-08-11:
This is better suited to ask at the [forum](https://forum.micropython.org) (and IIRC it has been asked and anwered there already) as it's merely a question, not an issue report or bug.

zxc199108, 2020-08-12:
OK ,Thanks

closed 2020-08-12 by zxc199108