micropython/micropython #6330: STM32F405 bootloader

MrZhouSX, 2020-08-12:
I am tring to add my own bootloader into the MicroPython project.
    I wrote a bootloader code myself, doing some of the work I needed. At the end of the code,I jump to the address of 0x08020000.then it start executing the code for MicroPython.
   Now I don't know how to compile MicroPython code.
   Have you done anything like this? Thanks!

davehylands, 2020-08-12:
This question would be better asked on the [forum](https://forum.micropython.org/)

The stm32 port has a bootloader called mboot which was intended for the Pyboard D series, but which also works on the pyboard. You'll need to change the memory layout in the linker script and may also need to change the filesystem location (depends on how big your bootloader is).

MrZhouSX, 2020-08-12:
Thanks!
The mboot must configure a pin to force entry into the boot loader ,and programming external SPI flash via the DFU interfaces.but,I don't want to use a extern pin and DFU.The workflow of my project is : Firstly, I load my bootloard into the MCU.Then,I transfer the MicroPython bin file over WIFI or BLE . So that I  can upgrade my projects freely through software.
Now,I just don't know how to compile MicroPython code.
My bootloader is about 9K ,
I am going to alloca bootloader from  0x08000000 to 0x08003fff about 16K /* sector 0 */
 filesystem from 0x08004000 to 0x0801ffff, LENGTH = 112K /* sectors 1,2,3,4  */
all the MicroPython text (contain the ISR and TEXT) from 0x08020000

davehylands, 2020-08-12:
Instructions for compiling MicroPython can be found in the README.md file: https://github.com/micropython/micropython#the-stm32-version

closed 2021-04-14 by dpgeorge