micropython/micropython #6332: Overflow converting -2147483648 to int when using MPZ

markstz, 2020-08-12:
When `MICROPY_LONGINT_IMPL == MICROPY_LONGINT_IMPL_MPZ` calling `mp_obj_get_int()` for a Python variable that contains the value `-2147483648` results in an `overflow converting long int to machine word` error.

If `MICROPY_LONGINT_IMPL == MICROPY_LONGINT_IMPL_LONGLONG` instead, the conversion works OK.

Is the MPZ overflow check incorrect?

dpgeorge, 2020-08-12:
> Is the MPZ overflow check incorrect?

The overflow check is done on the positive value first, before negating, so `2147483648=0x80000000` is overflowing a 32-bit signed integer.  It might make sense to fix this edge case so it does convert ok.

Can you explain the situation where this number needs to be converted to a machine word?

markstz, 2020-08-12:
It's an old unit test that runs a Python script and captures the return value from the script, converting it to a signed int using mp_obj_get_int(), and checks that all possible values work. When MICROPY_LONGINT_IMPL_LONGLONG was used the test passed, but when I switched to MICROPY_LONGINT_IMPL_MPZ it failed.