micropython/micropython #6333: porting fastLED

kdschlosser, 2020-08-12:
I am working on porting the fastLED library to MicroPython. While I am proficient in Python I am not in C/C++. If there is someone willing to lend a hand with porting the driver end of the code it would be greatly appreciated.

My goal is to keep as much of the API in tact as possible so anyone using fastLED would be able to convert their code to run in MicroPython fairly easily.

I am also curious to see how much interest there is in this being done. I also did want to mention that I would also be extending the port to include RGBW LED's

dlech, 2020-08-12:
I think the simplest thing to do would be to leave the FastLED code as-is. Then write a .cpp file that exposes what is needed as `extern "C"` (and fills in any missing Arduino functions, if needed). Then write a .c file that uses the new C API to create a MicroPython module.

kdschlosser, 2020-08-12:
If it was done that way wouldn't you need to use something like Cython to compile it because the code for FastLED is CPP???

dlech, 2020-08-12:
Yes, [perhaps something like this](https://github.com/micropython/micropython/pull/5083).

Or you could just compile it at the same time as the rest of MicroPython and create your own firmware.

kdschlosser, 2020-08-13:
That PR is about converting from .elf to .mpy not compiling an extension written in C/C++ to an extension that MicroPython would be able to load. Also because of all of the code in FastLED that is specific to directly accessing the hardware I think that there may be more issues with getting it to work then what it's worth.

There is another downfall. No support for 32 bit and above colors. only RGB. This is something people have been asking to get updated in Fast LED for 4 years and it still has not been done. This is something that can be done easily making this port. The reason I say it's easy is because we can pass an array of integers along with the bit depth to be used. The driver would use that bit depth to send out the correct number of bits. The order in which the bits need to be formatted would be done before passing the data to the driver. This way if a chipset gets extended like what was seen with the ws2812 where an extra color is added and the bit order gets changed even a user would be able to write the code needed to handle that and pass it to the driver to be sent.

I have a large portion of it already ported at this point (about 50%). The parts that would need to be written in C code are the drivers for the various LED chipsets. code that is similiar to what is seen in neopixel_write. This needs to be done in order to have the speed to keep the timings correct. From looking at the code for neopixel_write I cannot imagine it would be that difficult to write the drivers. I have looked at quite a few of the chipsets that are available and there are only small differences in how the data stream needs to be formatted. The largest difference between them is the hardware communication type/speed.

kdschlosser, 2020-08-17:
OK I made a repository for this port. The conversions I have done already have been committed and pushed already. If I had to put an amount on it I would say that I have converted between 75% and 80% of the code that needs to be converted to Python. I have not added support for RGBW just yet. Because FastLED heavily uses HSV adding RGBW support is going to be challenging. I have not been able to locate a conversion to convert from RGBW to RGB which is what would need to be done in order to utilize many of the "goodies" in FastLED. This is what I have found that discusses this kind of a conversion bit it only really dives into RGB to RGBW conversion and not the other way around.

https://stackoverflow.com/questions/21117842/converting-an-rgbw-color-to-a-standard-rgb-hsb-representation

I have not done to much digging as of yet and I am sure that I will come up with something once I get the library up and running and can run tests on it. There is going to need to be adjustments made to the RGB channels when the W channel so the color is not washed out. The problem with the above solution is that it uses min() and max() so there would be no way of knowing what the RGB values should be reverted to when converting from RGBW to RGB. Math that can be reversed is what needs to be used in order to be able to convert both ways. Possibly using percentages of each channel to determine the brightness needed for the white channel.

mbtronics, 2021-01-13:
Hi, I was looking at your micropython_fastled repo, but I have no idea how to compile/include it in micropython.
I can compiling micropython, including user modules, but that is with a micropython.mk make file.
Are there any instructions how to compile your fastled module?

kdschlosser, 2021-01-14:
@mbtronics 

open up an issue in my repo and I will discuss the library with you.

jonnor, 2024-07-02:
I believe this issue can be closed, as there is a dedicated repository linked - where any further discussions can be had.

closed 2024-07-04 by mattytrentini