micropython/micropython #6337: Lock up during garage collection on nrf52840-mdk-usb-dongle

v3xation, 2020-08-14:
I was hacking together a custom build off of Glenn's PR for LFS support and was having issues with periodic hard halt crashes. Eventually isolated it to garbage collection and then traced back to mainline. I have an old dirty version that works on these boards that I was using for low level radio support that worked so I did some sleuthing figuring this was a memory alignment problem. Noticed there is an .ld pushing the ram start off 8 bytes for open bootloader and wondered if the end was being calculated properly. Checked the map and it looks to be out of bounds:

firmware.map:8153:                0x0000000020040008                _ram_end = (ORIGIN (RAM) + LENGTH (RAM))

Solution implemented, add another tag to memory.ld: 
------------------------------------------------------------------------
_ram_offset    = DEFINED(_ram_offset) ? _ram_offset : 0;
RAM (xrw)       : ORIGIN = _app_ram_start, LENGTH = _app_ram_size - _ram_offset

and specify in nrf52840_open_bootloader.ld:    _ram_offset  = 0x08;

This appears to work and the end of ram is back at 0x20040000 where it should be, but I haven't investigated if I have knocked off alignment for anything else by implementing this change.

--Bug reproduction information--
Build command:
-----------------------------------------------------
 make BOARD=nrf52840-mdk-usb-dongle SD=s140 LTO=0

Compiler version:
------------------------------------------------------
arm-none-eabi-gcc (15:9-2019-q4-0ubuntu1) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

Prompt showing version and bug reproduction:
------------------------------------------------------
MicroPython v1.12-665-g60f5b941e on 2020-08-13; MDK-USB-DONGLE with NRF52840
Type "help()" for more information.
>>> import gc
>>> gc.collect()

jonnor, 2024-10-01:
Hi and thank you for the report. Is this problem still present in recent MicroPython, version 1.23 or later?

jonnor, 2025-03-03:
Missing information to reproduce, assuming fixed.

closed 2025-03-03 by jonnor