micropython/micropython #6338: Ampy stops the running program in esp 32 micro python

notgounded, 2020-08-14:
Not able use ampy and run python program at the same time . When  I use any ampy command it gives an error could not enter raw repl or it executives ampy command and stops the running program . How to use ampy command while ampy commands while program is running in esp wroom 32? Is it possible to run program on one core and use repl from other core?
Thanks in advance

davehylands, 2020-08-14:
This question is better suited to be asked on the forum https://forum.micropython.org/

github issues are for bugs.

I'm pretty sure that the answer to your question is no.

notgounded, 2020-08-14:
Thankyou and is there a way to use  2nd core in esp wroom 32?
Should I close this issue?

notgounded, 2020-08-15:
> This question is better suited to be asked on the forum https://forum.micropython.org/
> 
> github issues are for bugs.
> 
> I'm pretty sure that the answer to your question is no.

Thankyou and is there a way to use 2nd core in esp wroom 32?
Should I close this issue?

dpgeorge, 2020-08-26:
> . How to use ampy command while program is running in esp wroom 32? Is it possible to run program on one core and use repl from other core?

It's not possible to do this because MicroPython (on esp32 at least) does not have the concept of multiple interpreters running at once (one for the REPL and one for the running program).

Instead you could implement your own REPL in Python as a separate thread.

closed 2020-08-26 by dpgeorge