micropython/micropython #6340: Lacking support for multiple spi devices on the same bus

globalturist, 2020-08-15:
**Important: The platform Im working on is the Doit esp32 devkit v1.**

Iv'e come from a c++/Arduino background where if I want to use multiple spi devices on the same bus i just need to manage their slave select pin and use SPI.beginTransaction to set various speeds. However, in micropython there is no such thing as SPI.beginTransaction and I didn't find any custom drivers that ported the Arduino core SPI to micropython. Therefore, Im stuck with using init and deinit to imitate SPI.beginTransaction and SPI.endTransaction and the problem is it doesn't work. the sensors stop working completely.

### As for the reason I need such functionality:
I want to power an ili9341 TFT LCD and use both its touch and graphics capabilities and a MAX7219 seven segment display. In arduino I could just assign them the same spi bus(MISO, MOSI, SCK) for example vspi (23,19,18) and manually enable them through their slave select pin. However I dont see it possible with micropython.

Am I missing something obvious ? or does such a feature not exist in micropython ?

I can't attach .rar files for some reason so i will happily send anyone who asks the code.

mattytrentini, 2020-08-16:
I'm not sure I follow; if I understand you correctly, MicroPython operates as you describe. 
Connect multiple devices to the same SPI bus and use the CS line to select which device is active at any given time.

Perhaps the point of confusion is that, as the [MicroPython SPI documentation states](https://docs.micropython.org/en/latest/library/machine.SPI.html), control of the CS line is managed by the user. So you'll need to assign a Pin to each device to control their CS state.

chrismas9, 2020-08-16:
I use multiple SPI devices on STM32. I presume it would work on ESP32.

As well as asserting and de-asserting  the CS for each device you may have to re-init the clock phase and polarity to suit each peripheral.

There is an issue with STM32. I don't know if other ports have the same issue. When you change the clock polarity the init does not change the idle state on the clock line. This results in a missing clock edge at the start of the first byte transfer. To get around this I do a dummy write without any CS asserted.

globalturist, 2020-08-16:
Lets say I use 3 spi devices the first needs a 2.5mhz clock the second needs 40mhz and the third needs 100khz. In arduino i can put these 3 on the same bus and use SPI.beginTransaction(...) and just enable them through their cs pin when needed however if you try to initialize or re-initialize the same spi bus in micropython it doesn't seem to work escepcially when the only existing method to switch spi speeds is init->deinit.

chrismas9, 2020-08-16:
This may be ESP32 specific. The best thing to do is post the simplest script that demonstrates the problem, eg init, write byte, init different baud rate, write byte. Capture the SPI CLK pin with an oscilloscope. This will demonstrate the problem without requiring specific peripherals.

globalturist, 2020-08-16:
I cant attach files here so I'l point to the topic I opened on the micropython forum where there is 2 files attached that demonstrate it:
[https://forum.micropython.org/viewtopic.php?f=2&t=8887](url)

pumelo, 2020-08-16:
I'm having the same issue just that I did not know until I read this issue ... =)

Reading the source in `ports/esp32/machine_hw_spi.c` I think that there is exactly one supported configuration for each of the two spi buses at a time:

`// Static objects mapping to HSPI and VSPI hardware peripherals
STATIC machine_hw_spi_obj_t machine_hw_spi_obj[2];`

seems that micropython on esp32 does not know the concept of multiple devices for a bus with different polarity, baudrate ....

subsequent calls to 'SPI()' for the same bus will just overwrite the old configuration.

globalturist, 2020-08-16:
@pumelo Exactly! Thank you for pointing this out, you gave the best example for what i meant - short and to the point. Also, it makes the concept of SPI pretty much useless.

By the way did you also notice that the builtin bluetooth module doesn't even run on the ESP32 (it complains the module doesnt exist and the same goes for ubluetooth) and that the builtin SDCard module doesn't work as well which is oddish as it exists in the official micropython documentation.

globalturist, 2020-08-16:
@pumelo I almost forgot! If you read the source code you'll see there is no support for MUX pins thus you will never truely be able to run the SPI at 80 mhz....

pumelo, 2020-08-17:
@globalturist, I can't share complaints about SDCard or Bluetooth both work or worked fine for me (I'm not on the latest build though). In general I like like micropytho pretty much :heart_eyes:. May be you need to be more specific about the issues you have. 

Yesterday I modified ports/esp32/machine_spi_hw.c to work with multiple devices on the same bus which do have different polarity and phase configuration.

https://gist.github.com/pumelo/13fabe87e78532c105dd056e3c7198de

**This is a hack and WIP**, on first init you fix the pins which are used for the bus (miso, mosi, sck), thereafter you can't change theme. But you can have multiple devices on the same bus with different baudrate, polarity and phase configuration.

You need to exchange the given file and rebuild the fw. Let me know if it worked for you.

peterhinch, 2020-08-17:
I solved this problem on STM32 [for this display driver](https://github.com/peterhinch/micropython-epaper#spi-bus-arbitration). In this instance one of the devices was a Flash chip supporting a mounted filesystem, and the other was the display. The filesystem held glyphs for display, the two devices using the bus in conflicting ways. There was thus a lot of concurrency.

Perahps the same general approach would work on ESP32.

pumelo, 2020-08-17:
@peterhinch, thanks for your input. Does this imply, that the same behaviour is present in the stm32 port?

I'm aware that you can init the bus for each transaction to overcome this issue.

The origin of confusion is (from my POV), that you can init two spi buses without raising an exception but the first bus config is silently overwritten.

globalturist, 2020-08-18:
@pumelo 

> Yesterday I modified ports/esp32/machine_spi_hw.c to work with multiple devices on the same bus which do have different polarity and phase configuration.

https://gist.github.com/pumelo/13fabe87e78532c105dd056e3c7198de

This is a hack and WIP, on first init you fix the pins which are used for the bus (miso, mosi, sck), thereafter you can't change theme. But you can have multiple devices on the same bus with different baudrate, polarity and phase configuration.

You need to exchange the given file and rebuild the fw. Let me know if it worked for you.

Thanks for the help! I will definitely try it out (can't until next week as I don't have a pc with me)

> I'm aware that you can init the bus for each transaction to overcome this issue.

Tried using init and deinit technique on the esp32 and it doesn't work for some reason after the first deinit it stops working