micropython/micropython #6341: capta too hard

CarlFK, 2020-08-16:
I would leave a post about this on the forum, but I can't create an account because: 

https://forum.micropython.org/   
" you can see several micropython boards in the background of the  header image. What is the name of the I/O port between B9 and C1?"

I have no idea.  USB?  no. micro usb?  no.  i give up.

laurensvalk, 2020-08-17:
[image]

mcauser, 2020-08-17:
I believe that is a PyBoard 3 (PYBv3) in the forum header image.
Pinout can be found here:
https://github.com/micropython/pyboard/blob/master/pinout/pybv3/pinout.png

The pin between `B9` and `C1` is `C0`.

CarlFK, 2020-08-17:
C0 is correct.  c0 is not.
asking for "pin" would have helped, printing out it is case sensitive too.

closed 2021-04-14 by dpgeorge

geosmall, 2021-04-18:
Cute but not smart

ai4rv, 2021-12-23:
This is really stupid.  I had a hard time answering that, too--but it had nothing to do with "knowledge of hardware" or anything related to that.  I was because of the high-res monitor in my laptop, it is 4K but only 10: so sometimes I have to hold the thing up to my face to see anything!  Not to mention my eyes just aren't as good as they were....  At any rate, if I hadn't gargoyled that phrase, and someone hadn't posted this conversation in the forum, I wouldn't even be allowed to have an account on this forum, and for what reason?!  That just ticks me off.

okoeroo, 2022-01-08:
It took me this thread and an extra minute deciphering the messages here to have a mild clue on what is meant here. I know a few people whom would give up earlier on this. Can this be changed to have more humans on the forum?

tolap22, 2022-03-04:
Find the activation question a little annoying too, for the record the question for me was "**What is the name of the I/O port underneath A15?:** and the solution according to the png picture https://github.com/micropython/pyboard/blob/master/pinout/pybv3/pinout.png is "B3" all upper case!!

jouellnyc, 2022-06-30:
+1 - I only got to register b/c I googled and got here.

Oscer21, 2022-08-12:
Find the activation question a little annoying too, for the record the question for me was "What is the name of the I/O port underneath A15?: and the solution according to the png picture

[gbwhatsapp](https://www.apkguru.net/gbwhatsapp-apk/)

GBWhatsapp66, 2022-10-10:
GBWhatsApp is the modified version of the official WhatsApp. You get many extra features with gbwhatsapp instead of using the official WhatsApp[.](https://gbappsx.com/gbwhatsapp-apk/)