micropython/micropython #6342: stm32h7xx spi prescaler

koendv, 2020-08-16:
micropython 1.12 on stm32h743/h750. Configuring spi baudrate seems odd:
```
>>> from pyb import SPI
>>> spi2=SPI(2,SPI.MASTER,baudrate=1000000)
>>> spi2
SPI(2, SPI.MASTER, baudrate=24000000, prescaler=2, polarity=1, phase=0, bits=8)
```

This is STM32F7xx_HAL_Driver/Inc/stm32f7xx_hal_spi.h:
```
#define SPI_BAUDRATEPRESCALER_2         ((uint32_t)0x00000000U)
#define SPI_BAUDRATEPRESCALER_4         ((uint32_t)0x00000008U)
#define SPI_BAUDRATEPRESCALER_8         ((uint32_t)0x00000010U)
#define SPI_BAUDRATEPRESCALER_16        ((uint32_t)0x00000018U)
#define SPI_BAUDRATEPRESCALER_32        ((uint32_t)0x00000020U)
#define SPI_BAUDRATEPRESCALER_64        ((uint32_t)0x00000028U)
#define SPI_BAUDRATEPRESCALER_128       ((uint32_t)0x00000030U)
#define SPI_BAUDRATEPRESCALER_256       ((uint32_t)0x00000038U)
```
On stm32h7xx this becomes:
STM32H7xx_HAL_Driver/Inc/stm32h7xx_hal_spi.h:
```
#define SPI_BAUDRATEPRESCALER_2                       (0x00000000UL)
#define SPI_BAUDRATEPRESCALER_4                       (0x10000000UL)
#define SPI_BAUDRATEPRESCALER_8                       (0x20000000UL)
#define SPI_BAUDRATEPRESCALER_16                      (0x30000000UL)
#define SPI_BAUDRATEPRESCALER_32                      (0x40000000UL)
#define SPI_BAUDRATEPRESCALER_64                      (0x50000000UL)
#define SPI_BAUDRATEPRESCALER_128                     (0x60000000UL)
#define SPI_BAUDRATEPRESCALER_256                     (0x70000000UL)
```
as a result, the following line in ports/stm32/spi.c no longer works:
```
// compute baudrate
uint log_prescaler = (spi->Init.BaudRatePrescaler >> 3) + 1;
```
Suggest following patch:

```
diff --git a/ports/stm32/spi.c b/ports/stm32/spi.c
index 5f9b1c1a2..8a861556f 100644
--- a/ports/stm32/spi.c
+++ b/ports/stm32/spi.c
@@ -626,7 +626,24 @@ void spi_print(const mp_print_t *print, const spi_t *spi_obj, bool legacy) {
     if (spi->State != HAL_SPI_STATE_RESET) {
         if (spi->Init.Mode == SPI_MODE_MASTER) {
             // compute baudrate
-            uint log_prescaler = (spi->Init.BaudRatePrescaler >> 3) + 1;
+            uint log_prescaler = 1;
+            if (spi->Init.BaudRatePrescaler == SPI_BAUDRATEPRESCALER_2) {
+                log_prescaler = 1;
+            } else if (spi->Init.BaudRatePrescaler == SPI_BAUDRATEPRESCALER_4) {
+                log_prescaler = 2;
+            } else if (spi->Init.BaudRatePrescaler == SPI_BAUDRATEPRESCALER_8) {
+                log_prescaler = 3;
+            } else if (spi->Init.BaudRatePrescaler == SPI_BAUDRATEPRESCALER_16) {
+                log_prescaler = 4;
+            } else if (spi->Init.BaudRatePrescaler == SPI_BAUDRATEPRESCALER_32) {
+                log_prescaler = 5;
+            } else if (spi->Init.BaudRatePrescaler == SPI_BAUDRATEPRESCALER_64) {
+                log_prescaler = 6;
+            } else if (spi->Init.BaudRatePrescaler == SPI_BAUDRATEPRESCALER_128) {
+                log_prescaler = 7;
+            } else if (spi->Init.BaudRatePrescaler == SPI_BAUDRATEPRESCALER_256) {
+                log_prescaler = 8;
+            };
             uint baudrate = spi_get_source_freq(spi) >> log_prescaler;
             if (legacy) {
                 mp_printf(print, ", SPI.MASTER");
```
With this patch:
```
>>> spi2=SPI(2,SPI.MASTER,baudrate=1000000)
>>> spi2
SPI(2, SPI.MASTER, baudrate=750000, prescaler=64, polarity=1, phase=0, bits=8)
```
hth.

jcw, 2020-08-19:
... or `spi->Init.BaudRatePrescaler >> 28)` - when on STM32H7 (so an extra #define with the shift is all it takes).

dpgeorge, 2021-03-10:
Fixed by 79c186f5c962b514cde21415d7a262cfdbddbd97

closed 2021-03-10 by dpgeorge