micropython/micropython #6344: nRF microbit (nRF51822) issues

rcolistete, 2020-08-17:
I've built the ports/nrf for microbit with default configuration :
```
[nrf]$ make BOARD=microbit clean
[nrf]$ make -j8 BOARD=microbit
```
and using `$ screen /dev/ttyACM0 115200` to access REPL, but have some issues :
- **file system microbitfs is not accessible** via [`ufs`](https://microfs.readthedocs.io/en/latest/) or Mu Editor. How to externally acess the internal file system ? With official MicroPython for BBC micro:bit ufs and Mu Editor can access the files;
- `Ctrl+d` in REPL freezes, as well as `machine.soft_reset()`;
- `deepsleep()` and `lightsleep()` don't work. It seems that `__WFI`and `__WFE` are not working on microbit/nRF51822.

And building with Bluetooth :
```
[nrf]$ make BOARD=microbit clean
[nrf]$ make -j8 BOARD=microbit SD=s110
```
then REPL is not accessible at all via `$ screen /dev/ttyACM0 115200`.

Not issues, but features lacking with respect to [official MicroPython for BBC Micro:bit](https://github.com/bbcmicrobit/micropython) :
- no float numbers and math module, but it is possible to enable, see my [commit "nrf : nRF51 with float enabled"](https://github.com/rcolistete/micropython/commit/92294bb278882c297cf50696b168079eb3b737e4) in [my branch](https://github.com/rcolistete/micropython/tree/nrf51_float);
- module 'microbit' :
    *no `button_a.*`/`button_b.*`;
    * no `display.read_light_level()`;
    * no `accelerometer.*`;
    * no `compass.*`.

About (1), it is simple to read via `machine.Pin` :
```
from machine import Pin
button_a = Pin(Pin.cpu.P17, Pin.IN, Pin.PULL_UP)
button_b = Pin(Pin.cpu.P26, Pin.IN, Pin.PULL_UP)
```

rcolistete, 2020-08-19:
1st issue is caused by 2nd issue !

Workaround for 1st issue, "file system microbitfs is not accessible via ufs or Mu Editor" :
- ufs (microfs) alone and used by Mu Editor does a soft reset by sending a "Ctrl+d" command, which on MicroPytho mainline on nRF51822/BBC micro:bit currently has a bug that freezes;
- solution is to coment [lines 93-94 of 'microfs/microfs.py'](https://github.com/ntoll/microfs/blob/master/microfs.py#L93);
- after turning on (boot) BBC Micro:bit, press reset button of BBC Micro:bit so ufs/Mu Editor->Files can work.

Of course it would be best to solve the `machine.soft_reset()` bug.

rcolistete, 2020-08-20:
One more issue on BBC Micro:bit, unstable I2C scan reading :
```
from machine import Pin, I2C
i2c = I2C(0, scl=Pin.board.I2C_SCL, sda=Pin.board.I2C_SDA) 
i2c.scan() 
    [14, 25, 30, 43, 73]
i2c.scan() 
    [13, 25, 30, 87]
i2c.scan() 
    [13, 25, 30, 43, 88, 102, 107]
i2c.scan() 
    [19, 25, 30, 49]
i2c.scan() 
    [18, 25, 30, 40, 70, 84, 99, 115]
```
It should return the same I2C addresses.

pvonmoradi, 2020-12-03:
@rcolistete Doesn't ufs specifically check for a Micro:Bit device? Is it possible to use if for nRF51/52 chips flashed with mbfs?