micropython/micropython #6345: idea: host MPY as front-end for attached board

jcw, 2020-08-19:
MPY builds bytecode on-the-fly when in interactive mode:

```txt
$ micropython -v -v
MicroPython v1.12-483-g22806ed5d-dirty on 2020-05-29; darwin version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> print(1+2)
File <stdin>, code block '<module>' (descriptor: 7f9b5f506140, bytecode @7f9b5f506900 23 bytes)
Raw bytecode (code_info_size=7, bytecode_size=16):
 10 0a 07 00 ea 00 00 11 cd 00 00 11 7b 00 00 83
 34 01 34 01 59 51 63
arg names:
(N_STATE 3)
(N_EXC_STACK 0)
  bc=0 line=1
00 LOAD_NAME __repl_print__ (cache=0)
04 LOAD_NAME print (cache=0)
08 LOAD_CONST_SMALL_INT 3
09 CALL_FUNCTION n=1 nkw=0
11 CALL_FUNCTION n=1 nkw=0
13 POP_TOP
14 LOAD_CONST_NONE
15 RETURN_VALUE
3
>>>
```

Would it be an idea to extend the unix build to directly connect to a remote board over serial, so that MPY becomes an interactive front-end, with all the conveniences of staying on the host for development? As extra benefit, this would then also work with low-end boards which don't have the memory to support the compiler on-board.

What this would involve, I suspect - apart from the actual serial connection details - is that MPY needs to generate a complete module-like package, à la `mpy-cross`, with qstrings and constants and send that over.

dpgeorge, 2020-08-26:
This would be a lot of work / a lot of changes.

As an alternative see #6375.  It allows a similar workflow as proposed above, to have all files located on the PC.  It could be made to support a board without a compiler by precompiling everything to `.mpy` files on the PC, and then the board imports those `.mpy` files directly via the serial connection.

jcw, 2020-08-26:
Ok, clear, thanks - `mpr` looks like an interesting option. Not quite as interactive, but still.

closed 2020-08-26 by jcw

dpgeorge, 2020-08-26:
>  Not quite as interactive, but still.

One workflow you can use with `mpr` is:
1. chdir on your host PC to the development directory
2. `mpr mount . repl` to get a prompt on the board, with PC filesystem active
3. edit (and possibly mpy-cross compile) your code on the PC with your favourite editor/IDE
4. ctrl-D on the board to refresh, then `import main` to execute your application
5. ctrl-C to stop the application and inspect it
6. go to 3