micropython/micropython #6347: `random.choice()` returning the ever same values and sequences

mirko, 2020-08-20:
On my ESP32 - despite of what I/it did befpre (incl. network connectivity, keyboard input, etc.) - `random.choice()` always returns the ever same values and sequences:

```
>>> from random import choice
>>> choice('01234567890')
'6'
>>> choice('01234567890')
'8'
>>> choice('01234567890')
'3'
>>> print("foobar")
foobar
>>> choice('01234567890')
'3'
>>> choice('01234567890')
'5'
>>> from machine import reset
>>> reset()
[..]
MicroPython v1.12-170-g43c91bea2-dirty on 2020-08-10; FOOBAR DEBUG BUILD with ESP32
Type "help()" for more information.
>>> from random import choice
>>> choice('01234567890')
'6'
>>> choice('01234567890')
'8'
>>> choice('01234567890')
'3'
>>> print("SOMETHING ELSE")
SOMETHING ELSE
>>> choice('01234567890')
'3'
>>> choice('01234567890')
'5'
>>>
```

Apparently this doesn't only happen on the very same device, but the sequence is identical across devices.
I was even told that holds true for every port (if implemented I suppose, which e.g. is not the case for the unix-port, so I couldn't test/verify).

I was also pointed towards `random.seed()` which indeed changes the behaviour of `random.choice()` once called.

Also, I was being made aware of this being "legitimate" behaviour, as CPython handles things the same (although, `random.seed()` does not /require/ a seed as arg, while mpy does).

Even if all this is the case (which I have no reason to doubt):

This is... at least unexpected and I'd even generalise and say rather counter-intuitive. When calling a function called random (or being part of such a module), explicitly designed to return something random, and I do not get anything random but can't even tell so (at first glance at least) and then use it in code which quite likely /relies/ on it being random (collision avoidance, crypto, etc.), then I'd say the risk of people running into this and shooting their feet way worse than I just did, is significantly high.

robert-hh, 2020-08-20:
The behavior which you have seen is typical for  a pseudo-RNG. What me puzzles here is that the documentation about the random module is missing. That may contain a warning like it is done for uos.urandom(). Besides that, the ESP32 has a hardware RNG, which is active when WifI or Bluetooth is used. I did not look into the source code, how is generates random numbers. Using the hardware RNG may be an option. But even then, one has to cope for stiuations when neither WiFi nor Bluetooth are used.

dlech, 2020-08-20:
> the documentation about the random module is missing

There is already an issue open for this #5518

mirko, 2020-08-20:
> > the documentation about the random module is missing
> 
> There is already an issue open for this #5518

Ok, that's about the docs - but the behaviour - in my view - is still questionable.

mirko, 2020-08-20:
From a quick test CPython's implementation at least returns different - seemingly random - sequences across interpreter instances.

jimmo, 2020-08-24:
I think in the situation where you have a HW RNG (or any source of randomness really), @robert-hh 's PR (#6362) makes sense.
If you want deterministic simulations and random is just a source of numbers, then explicitly seeding is what you'd need to do in CPython anyway.

So that mostly just leaves MCUs that don't have a HW RNG (or other convenient source of entropy). I think it's a dangerous game to start getting into inventing our own entropy source (e.g. temperature, ADCs, etc) (not to mention a lot of code), but I guess there are some easy answers to at least make it different across different chips (even if the sequence is the same per-chip) (@mirko suggests using the same source that mp_hal_get_mac uses on STM32 to generate an LAA MAC).

As I mentioned on #6362 I'd like to see a no-arg version of `random.seed()`.

Alternatively we can document the random module to describe exactly the way it works, and make it clear to the user if they want HW randomness, then they need to be on a support board, and they can explicitly use  `random.seed(os.urandom(4))` (which we would need to make work, as it does on CPython).

mirko, 2020-08-24:
From what I take out of it so far is, no matter which way we go, the module `random` is not to be generally trusted to return true randomness.
Even if we go with a full-blown default pre-seed which might involve a true RNG, you'd need still to know whether the platform you're on and the way it's configured/used (e.g. wifi/ble on/off) provides the `random`-module with true randomness - or not.

/Still/ I think it makes sense to pre-seed by default, in order to be able to generate (fairly) unique (U)IDs - across interpreter restarts and, more importantly, across devices (otherwise I feel like it's just a matter of time until people run into the backend/broker/server.. receiving the same sequence of supposed to be unique IDs - from different devices). Here, as @jimmo already said, I'd suggest using (at least) time and some sort of machine-id (CPU/MAC/..) and then maybe encourage the user to further seed it with more data once available (e.g. received via network, etc.).

(Does that actually work? Can I call `seed()` multiple times, /adding/ new entropy? I'm coming from unix, where I'm used to write stuff into `/dev/random` which didn't already announce itself as entropy source on kernel level).

I'd see (and document) the `random` module then as sth. alike `/dev/urandom` in UNIX/Linux - providing a constant stream of "randomness" at all times (however its actual entropy solely depends on its seed-source(s)) and `uos.random` as sth. alike `/dev/random` (well, if that's actually the case(?)) providing limited, blocking but true(TM) randomness.

No matter how things end up: as long as I understand the whole thing eventually, I'm happy and offering myself to write/enhance docs about it in the end.

peterhinch, 2020-08-24:
@mirko In networked applications can you not derive a node-unique number from the MAC address? You can use this to derive unique ID's or as a seed.

robert-hh, 2020-08-24:
@mirko 
> Can I call seed() multiple times, /adding/ new entropy? I'm coming from unix, where I'm used to write stuff into /dev/random which didn't already announce itself as entropy source on kernel level).

The microypython implementation of random uses as pseudo-RNG. That can be seeded as often as you like and will create with the same seed the same sequence of numbers.

mirko, 2020-08-24:
> @mirko In networked applications can you not derive a node-unique number from the MAC address? You can use this to derive unique ID's or as a seed.

@peterhinch Isn't this this what I tried to say by
> I'd suggest using (at least) time and some sort of machine-id (CPU/MAC/..)
or what @jimmo said by
> @mirko suggests using the same source that mp_hal_get_mac uses on STM32 to generate an LAA MAC
or am I missing sth.?

mirko, 2020-08-24:
> @mirko
>
>
> The microypython implementation of random uses as pseudo-RNG. That can be seeded as often as you like and will create with the same seed the same sequence of numbers.

Just to make sure, it /adds/ to the entropy pool or it re-seeds - as in - resets with the new value?

robert-hh, 2020-08-24:
It re-seeds. The internal state of the RNG will completely set to a defined value. That is intentional. Otherwise the (pseudo-)RNG would not generate the same sequence again. 
On a side note: I made a test how many bits of the first random number are changed between two seeds which differ by one bit. It's about 4 bit change when requesting 30 bit numbers with getrandbits().

peterhinch, 2020-08-25:
@mirko My point was that the use-case of constructing a network-unique machine ID from the MAC address can be met in Python with current firmware. I did a lousy job of expressing this. Apologies.

closed 2020-10-29 by dpgeorge

mirko, 2020-11-07:
Having a best practice question: so, random is pre-seeded with values from an RNG - if platforms provides such (correct?)
What if I'm having a additional source for entropy and would like to /add/ this to a seed.
Since I can only re-seed the PRNG as far as I understood, what's the best way to re-seed the PRNG with the (platform specific) RNG /and/ values I do have from other sources?
Would it be `random.seed(RANDOM_STR_FROM_SENSOR1 + RANDOM_STR_FROM_WHATEVER + HW-RNG)` or is there anything/better other than that?

robert-hh, 2020-11-07:
When you combine values from several sources for seeding, like in the example above, you should XOR them. That ensure better mixing.
uos.urandom() uses a hardware RNG, whenever available, while the urandom() modules uses a pseudo-RNG. So the two may not be related at all.

mirko, 2020-11-07:
> When you combine values from several sources for seeding, like in the example above, you should XOR them. That ensure better mixing.
> uos.urandom() uses a hardware RNG, whenever available, while the urandom() modules uses a pseudo-RNG. So the two may not be related at all.

Thanks! Just for clarification: what should I call to get values from the actual RNG to then XOR with my other sources and pass it to `seed()`?

robert-hh, 2020-11-07:
That depends on which board you are using. The PyBoard seem to have a hardware RNG, the ESP32 and ESP8266 also claim to have one. If yes, then you can use uos.urandom() as source. If in doubt, look into the respecitive implementation uos.urandom() in moduos.c

PR #6558 (2020-10-29): extmod: Seed random module on import (v2)
This PR builds on #6362.  It contains:
- random module is seeded with hardware RNG on stm32, esp8266, esp32
- for sm32 MCUs without hardware RNG, the seeding of the pRNG is improved, uses systick+rtc+unique-id
- adds support for `random.seed()` and `random.seed(None)`, which re-seeds the module with a hardware RNG
- `import random` calls `random.seed()` (instead of passing in the HW RNG to `seed(...)`)
- fixes a bug where the seed is initialised twice if the module is imported by different names, eg `random` and `urandom`

Fixes #6347.