micropython/micropython #6354: bluetooth "GAP procedure initiated ..." messages

teuler, 2020-08-21:
Hi,
is there a way to prevent these messages being automatically printed to the REPL?
Thanks

dpgeorge, 2020-08-26:
You can use the `esp.osdebug(None)` function.

teuler, 2020-08-31:
I have this in `boot.py`. It does not prevent these "GAP procedure ..." messages from being printed to the REPL. Also, if I call `esp.osdebug(None)` directly before initiating the BLE link, the messages still appear.

jimmo, 2020-09-10:
I took a quick look at this... quick summary:
ESP32's NimBLE port is set up to use the ESP-IDF logging functions, which is controlled by `esp.osdebug(None)`, which will globally set the log level to `ERROR` and clear the level on any tag. This particular message is printed at level `INFO`, and so `esp.osdebug(None)` does prevent this message from being logged.

However... for reasons that aren't clear to me, inside the IDF's `nimble_port.c`, they explicitly `esp_log_level_set("NimBLE", LOG_LOCAL_LEVEL);`, which has the effect of setting the level for the NimBLE tag back to `INFO`. This function is called by `ble.active(True)`.

So the simple workaround is to call `esp.osdebug(None)` after `ble.active(True)`. I can't see a better workaround other than to change `CONFIG_LOG_LOCAL_LEVEL` to `ERROR` in MicroPython builds (i.e. in `ports/esp32/boards/sdkconfig.base`). This might also help with other reports of spurious logging on ESP32). I will send a PR for this for discussion.

teuler, 2020-09-12:
Thanks for looking into that,

dpgeorge, 2021-04-14:
I haven't seen these messages since the move to cmake and newer IDF versions like v4.2.  So I'll close this issue.

closed 2021-04-14 by dpgeorge