micropython/micropython #6366: no \_\_dict\_\_ attribute in a module object

kdschlosser, 2020-08-24:
chipset: ESP32
MicroPython version: 1.12

Traceback: ```python
Traceback (most recent call last):
  File "boot.py", line 1, in <module>
  File "LEDS/__init__.py", line 134, in <module>
  File "LEDS/__init__.py", line 115, in __init__
AttributeError: 'module' object has no attribute '__dict__'
```

Code to reproduce error: ```python
import sys

TEST_VARIABLE = 'This is a test'

print(sys.modules[__name__].__dict__)
```

Run the code above in both CPython and MicroPython, you will not get an error when it runs in CPython

vshymanskyy, 2024-09-23:
Cannot reproduce this. Probably safe to close

dpgeorge, 2024-09-23:
Support was added for module `__dict__` in commit 15d0615d5cad430930d160a8ca3e809d7b82da32 (September 2022).

So yes we can close this!

closed 2024-09-23 by dpgeorge

kdschlosser, 2024-09-23:
Yes thanks.