micropython/micropython #6367: object class does not have __setattr__ function

kdschlosser, 2020-08-24:
Chipset: ESP32
MicroPython version: 1.12

Traceback:
```python
Traceback (most recent call last):
  File "boot.py", line 1, in <module>
  File "LEDS/__init__.py", line 144, in <module>
  File "LEDS/__init__.py", line 119, in __init__
AttributeError: type object 'object' has no attribute '__setattr__'
```

Code to replicate problem:
```python

class SomeClass(object):
    
    def __init__(self):
        self._test = 'testing 123'

    def __setattr__(self, key, value):
        if key.startswith('_'):
            object.__setattr__(self, key, value)
        else:
            raise AttributeError('only private attributes allowed')

some_class = SomeClass()
```

Run the above coe in CPython and then in MicroPython, you will see there is no error in CPython.

jimmo, 2020-08-24:
This was implemented in #5404 which was merged just after v1.12 was released (so it'll be in the upcoming v1.13, which should be quite soon).

kdschlosser, 2020-08-24:
@jimmo 

is just \_\_setattr\_\_ added or are the rest of the missing magic methods also added?...

I know in the documentation it states that there is limited functionality of the data types. I haven't really dug that deep to see what is missing. I know that \_\_dir\_\_ is missing as well.

Also is there a way to get an ESP to stop boot looping because of a kernel panic so that new scripts can be uploaded to try and isolate which script file is causing the kernel panic without having to flash MicroPython each time?

peterhinch, 2020-08-25:
> Also is there a way to get an ESP to stop boot looping

Start your scripts at the REPL rather than from main.py?

kdschlosser, 2020-08-26:
@peterhinch 

DOAH! I didn't think of that, don't know why I didn't but I didn't.

Thanks for helping with the boot loop issue. I still don't know what in my code could cause a kernel panic. and this is just on importing the files I don't even get to the point to run the program before the kernel panic happens. I need to isolate the file that is causing the panic

closed 2025-03-03 by kdschlosser