micropython/micropython #6368: (ESP32 uPython V1.12) List of List: Bad assignement

FranklinBerger, 2020-08-24:
Hello,
I have found that bug when dealing with lists. The problem is that when you try to creat a list from multiplications, they all become linked together. I have tried to explain the bug on the most simple way via the screenshot. Hope it will result on a bug fix or something like that.
I'm on a ESP32 with Micropython esp32-idf3-20191220-v1.12
Thank you for everything!

[image]

jcw, 2020-08-24:
See https://www.tutorialspoint.com/How-does-repetition-operator-work-on-list-in-Python

Try `a = [[0] * 5 for _ in range(3)]`

FranklinBerger, 2020-08-24:
Oh, don't expected that but it work, thank you verry much

closed 2020-08-24 by FranklinBerger

davehylands, 2020-08-24:
You get the same result using CPython, so it isn't actually a bug.

FranklinBerger, 2020-08-25:
Yes, I saw that after,
A friend sent me that post on StackOverflow witch explain everthing:

https://stackoverflow.com/questions/12791501/python-initializing-a-list-of-lists

It just seems so stange that I didn't thought it would be a normal way of working for Python / uPython.
Thank you for you'r kind responses and sorry for this "trash-issue"