micropython/micropython #6372: nrf52840 dk could not enter raw repl (Zephyr Port)

mridwang, 2020-08-25:
I am facing problem in copying the python script to the board. I am using the method given in the following link
http://docs.micropython.org/en/v1.12/reference/pyboard.py.html but I am getting the following error. Please help me. I will be grateful.

python3 pyboard.py --device /dev/ttyACM0 light.py
b'>>> \r\n>'
could not enter raw repl

I have tried ampy and rshell as well and I am unable to enter raw repl

Regards

dpgeorge, 2020-08-26:
Using zephyr 2.3.0 with latest MicroPython on an `nrf52dk_nrf52832` board, running scripts via `pyboard.py script.py` works for me.

But note that there is no filesystem on the nrf52 boards (by default) so you will not be able to copy files to it.

To enable a filesystem you need to build with `CONFIG_FLASH=y` and `CONFIG_FLASH_MAP=y` (at least).  See commits 86a66960f9a3538de9d48abb719975741569495f and 7a5752a7489f6be1c7307455b33119888392a09d

mridwang, 2020-08-26:
Sir, thank you very much for your response. I have tried the configurations by enabling only "CONFIG_FLASH_MAP=y" and in combination with "CONFIG_DISK_ACCESS_SDHC=y" and also with only CONFIG_FLASH=y and CONFIG_FLASH_MAP=y . With the aforementioned config, the program completes the MAKE execution but cannot enter raw REPL.

I am running the command "...../micropython/tools $ python3 pyboard.py --device /dev/ttyACM0 light.py" (/micropython/tools....where pyboard.py and light.py (my application) is residing). 
 
python3 pyboard.py --device /dev/ttyACM0 light.py
b'>>> \r\n>'
could not enter raw repl

I have tried on latest version of Micropython and Zephyr 2.3.0. 
Please advise.
Thanks in advance.

mridwang, 2020-08-26:
Moreover, when I add in "**nrf52840dk_nrf52840.conf**" the following configurations: 
**CONFIG_FLASH=y
CONFIG_FLASH_MAP=y
CONFIG_MPU_ALLOW_FLASH_WRITE=y**

I get the following error during the execution of "**make BOARD=nrf52840dk_nrf52840**"

make BOARD=nrf52840dk_nrf52840
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
Warning: FROZEN_DIR is deprecated in favour of FROZEN_MANIFEST
mkdir -p outdir/nrf52840dk_nrf52840 && cmake -DBOARD=nrf52840dk_nrf52840  -Boutdir/nrf52840dk_nrf52840 -H.
Including boilerplate (Zephyr workspace): /home/rg/zephyrproject/zephyr/cmake/app/boilerplate.cmake
-- Application: /home/rg/zephyrproject/micropython-master/ports/zephyr
-- Zephyr version: 2.3.0 (/home/rg/zephyrproject/zephyr)
-- Found Python3: /usr/bin/python3.8 (found suitable exact version "3.8.2") found components: Interpreter 
-- Board: nrf52840dk_nrf52840
ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found toolchain: zephyr (/home/rg/zephyr-sdk-0.11.3)
-- Found west: /home/rg/.local/bin/west (found suitable version "0.7.3", minimum required is "0.7.1")
-- Found dtc: /home/rg/zephyr-sdk-0.11.3/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found BOARD.dts: /home/rg/zephyrproject/zephyr/boards/arm/nrf52840dk_nrf52840/nrf52840dk_nrf52840.dts
-- Generated zephyr.dts: /home/rg/zephyrproject/micropython-master/ports/zephyr/outdir/nrf52840dk_nrf52840/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /home/rg/zephyrproject/micropython-master/ports/zephyr/outdir/nrf52840dk_nrf52840/zephyr/include/generated/devicetree_unfixed.h
Parsing /home/rg/zephyrproject/zephyr/Kconfig
Loaded configuration '/home/rg/zephyrproject/zephyr/boards/arm/nrf52840dk_nrf52840/nrf52840dk_nrf52840_defconfig'
Merged configuration '/home/rg/zephyrproject/micropython-master/ports/zephyr/prj.conf'
Merged configuration '/home/rg/zephyrproject/micropython-master/ports/zephyr/boards/nrf52840dk_nrf52840.conf'
Configuration saved to '/home/rg/zephyrproject/micropython-master/ports/zephyr/outdir/nrf52840dk_nrf52840/zephyr/.config'
Kconfig header saved to '/home/rg/zephyrproject/micropython-master/ports/zephyr/outdir/nrf52840dk_nrf52840/zephyr/include/generated/autoconf.h'

warning: TEST_RANDOM_GENERATOR (defined at boards/shields/esp_8266/boards/sam4e_xpro.defconfig:17,
subsys/random/Kconfig:8) was assigned the value 'y' but got the value 'n'. Check these unsatisfied
dependencies: ((BOARD_SAM4E_XPRO && NETWORKING && SHIELD_ESP_8266) || !ENTROPY_HAS_DRIVER) (=n). See
http://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_TEST_RANDOM_GENERATOR.html and/or look
up TEST_RANDOM_GENERATOR in the menuconfig/guiconfig interface. The Application Development Primer,
Setting Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be
helpful too.

-- The C compiler identification is GNU 9.2.0
-- The CXX compiler identification is GNU 9.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/rg/zephyr-sdk-0.11.3/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
-- Cache files will be written to: /home/rg/.cache/zephyr
-- Configuring done
-- Generating done
-- Build files have been written to: /home/rg/zephyrproject/micropython-master/ports/zephyr/outdir/nrf52840dk_nrf52840
make --no-print-directory -C outdir/nrf52840dk_nrf52840 outputexports CMAKE_COMMAND=: >outdir/nrf52840dk_nrf52840/Makefile.export
make -C outdir/nrf52840dk_nrf52840 syscall_list_h_target kobj_types_h_target
make[1]: Entering directory '/home/rg/zephyrproject/micropython-master/ports/zephyr/outdir/nrf52840dk_nrf52840'
make[2]: Entering directory '/home/rg/zephyrproject/micropython-master/ports/zephyr/outdir/nrf52840dk_nrf52840'
make[3]: Entering directory '/home/rg/zephyrproject/micropython-master/ports/zephyr/outdir/nrf52840dk_nrf52840'
make[4]: Entering directory '/home/rg/zephyrproject/micropython-master/ports/zephyr/outdir/nrf52840dk_nrf52840'
Scanning dependencies of target parse_syscalls_target
make[4]: Leaving directory '/home/rg/zephyrproject/micropython-master/ports/zephyr/outdir/nrf52840dk_nrf52840'
make[4]: Entering directory '/home/rg/zephyrproject/micropython-master/ports/zephyr/outdir/nrf52840dk_nrf52840'
[  0%] Generating misc/generated/syscalls.json, misc/generated/struct_tags.json
make[4]: Leaving directory '/home/rg/zephyrproject/micropython-master/ports/zephyr/outdir/nrf52840dk_nrf52840'
[ 33%] Built target parse_syscalls_target
make[4]: Entering directory '/home/rg/zephyrproject/micropython-master/ports/zephyr/outdir/nrf52840dk_nrf52840'
Scanning dependencies of target syscall_list_h_target
make[4]: Leaving directory '/home/rg/zephyrproject/micropython-master/ports/zephyr/outdir/nrf52840dk_nrf52840'
make[4]: Entering directory '/home/rg/zephyrproject/micropython-master/ports/zephyr/outdir/nrf52840dk_nrf52840'
[ 33%] Generating include/generated/syscall_dispatch.c, include/generated/syscall_list.h
make[4]: Leaving directory '/home/rg/zephyrproject/micropython-master/ports/zephyr/outdir/nrf52840dk_nrf52840'
[100%] Built target syscall_list_h_target
make[3]: Leaving directory '/home/rg/zephyrproject/micropython-master/ports/zephyr/outdir/nrf52840dk_nrf52840'
make[2]: Leaving directory '/home/rg/zephyrproject/micropython-master/ports/zephyr/outdir/nrf52840dk_nrf52840'
make[2]: Entering directory '/home/rg/zephyrproject/micropython-master/ports/zephyr/outdir/nrf52840dk_nrf52840'
[... cut ...]
   83 |     { MP_ROM_QSTR(MP_QSTR_FlashArea), MP_ROM_PTR(&zephyr_flash_area_type) },
      |                   ^~~~~~~~~~~~~~~~~
../../py/obj.h:96:56: note: in definition of macro 'MP_OBJ_NEW_QSTR'
   96 | #define MP_OBJ_NEW_QSTR(qst) ((mp_obj_t)((((mp_uint_t)(qst)) << 3) | 2))
      |                                                        ^~~
modzephyr.c:83:7: note: in expansion of macro 'MP_ROM_QSTR'
   83 |     { MP_ROM_QSTR(MP_QSTR_FlashArea), MP_ROM_PTR(&zephyr_flash_area_type) },
      |       ^~~~~~~~~~~
make: *** [../../py/mkrules.mk:63: build/modzephyr.o] Error 1

dpgeorge, 2020-08-26:
try a `make clean` first

mridwang, 2020-08-26:
Thanks sir. Previously I was only deleting the folder from OUTDIR. Your suggestion worked, now I am able to execute the **"make BOARD=nrf52840dk_nrf52840"** by including following configurations. 
# Flash drivers
CONFIG_FLASH=y
CONFIG_FLASH_MAP=y
CONFIG_MPU_ALLOW_FLASH_WRITE=y

But still I am an unable to enter the RAW REPL and getting the following error....
~/zephyrproject/micropython-master/tools$ python3 pyboard.py --device /dev/ttyACM0 light.py 
b'\r'
~/zephyrproject/micropython-master/tools$ python3 pyboard.py --device /dev/ttyACM0 light.py 
b'\r'
~/zephyrproject/micropython-master/tools$ python3 pyboard.py --device /dev/ttyACM0 light.py 
b'\r'
could not enter raw repl

mridwang, 2020-08-26:
Sir, I have tried on another nrf52840 dk, I could not enter raw REPL. 
Please advise. I will be grateful. 
Regards

dpgeorge, 2020-08-26:
As I mentioned above, it worked ok for me on a pca10040.

To debug try the following:
1. open the port with a serial/terminal program like picocom
2. enter ctrl-C followed by ctrl-B; it should print the banner and give a `>>> ` prompt
3. enter ctrl-A; it should say `raw REPL; CTRL-B to exit`
4. enter ctrl-D; it should say `MPY: soft reboot` then `raw REPL; CTRL-B to exit`
5. enter `print(1)` then ctrl-D; it shou/d say `OK1`
6. enter ctrl-B; it should print the banner and be back at the `>>> ` prompt

Please paste all output of the above commands here to help debug.

glennrub, 2020-08-26:
@mridwang , i'm not sure if it helps, but you could try to turn off the mass storage device, if it is enabled. I have a feeling i have experienced the same as you earlier, but with the nrf-port, not the zephyr-port.

To disable Mass Storage Device, connect with JLink commander (JLinkExe in Linux). 
```
JLinkExe -if swd -device nrf52832_xxaa -speed 4000
```
Then issue `MSDDisable` and quit, then power cycle the device.
```
J-Link>MSDDisable
Probe configured successfully.
J-Link>q
```
You should be able to enable it again by issuing `MSDEnable`.

mridwang, 2020-08-27:
@dpgeorge 

**1.** I opened the Serial Port with Putty.

**2.** MicroPython v1.12 on 2020-08-27; zephyr-nrf52840dk_nrf52840 with nRF52840_QIAA
Type "help()" for more information.
**2.** >>>
**3.** raw REPL; CTRL-B to exit
>
MicroPython v1.12 on 2020-08-27; zephyr-nrf52840dk_nrf52840 with nRF52840_QIAA
Type "help()" for more information.
>>>
**4.** soft reboot
MicroPython v1.12 on 2020-08-27; zephyr-nrf52840dk_nrf52840 with nRF52840_QIAA
Type "help()" for more information.
>>>
MicroPython v1.12 on 2020-08-27; zephyr-nrf52840dk_nrf52840 with nRF52840_QIAA
Type "help()" for more information.
**5.** >>> print(1)
1
>>>
soft reboot
MicroPython v1.12 on 2020-08-27; zephyr-nrf52840dk_nrf52840 with nRF52840_QIAA
Type "help()" for more information.
>>>
**6.** MicroPython v1.12 on 2020-08-27; zephyr-nrf52840dk_nrf52840 with nRF52840_QIAA
Type "help()" for more information.
>>>

For Point 4. it did not enter the Raw REPL. 
For Point 5. it did not say 0K1, after Ctrl D it did the soft boot again. 

Please advise. 

Thank you

mridwang, 2020-08-27:
@glennrub

Thank you for your input. I have tried it as per your suggestion but getting the same error.

dpgeorge, 2020-08-27:
That's very strange that it cannot enter raw REPL by manually typing ctrl-A... if you just type ctrl-C and then ctrl-A, what exactly is the output?

It may be putty that is doing something strange.  Try a different terminal program like tera term.

mridwang, 2020-08-27:
**1. When I press ctrl-C, it gives the following output:**
 >>>
**2. When I press ctrl-A, it gives the following output:** 
raw REPL; CTRL-B to exit

Combining both outputs: 
>>>
raw REPL; CTRL-B to exit

dpgeorge, 2020-08-27:
ok, what happens when you start from fresh (ie hard reset) press ctrl-C, then ctrl-A, then ctrl-D?

mridwang, 2020-08-27:
_>>>_
raw REPL; CTRL-B to exit
_>_OK
soft reboot
raw REPL; CTRL-B to exit
_>_

dpgeorge, 2020-08-27:
That output is different to what you reported above in https://github.com/micropython/micropython/issues/6372#issuecomment-681282243

So it can enter the raw REPL.  Now try entering `print(1)` then ctrl-D, and it should print `OK1`.

mridwang, 2020-08-27:
_>>>_
raw REPL; CTRL-B to exit
_>_OK
soft reboot
raw REPL; CTRL-B to exit
_>_
MicroPython v1.12 on 2020-08-27; zephyr-nrf52840dk_nrf52840 with nRF52840_QIAA
Type "help()" for more information.
_>>>_ print(1)
1
_>>>_
soft reboot
MicroPython v1.12 on 2020-08-27; zephyr-nrf52840dk_nrf52840 with nRF52840_QIAA
Type "help()" for more information.
_>>>__

mridwang, 2020-08-27:
I have tried it on picocom as well

dpgeorge, 2020-08-27:
> It is not giving 0K1

I can see... there is something wrong... do you have a different board (with different MCU) you can try, to see if it's an issue with the board or PC side?

mridwang, 2020-08-27:
Sir, yesterday I tried different nrf52840 dk and today I am using the third one. But I did not change the Personal Computer. May I try it on the other PC as well? For your information I am using Ubuntu 20.04.1 LTS.

dpgeorge, 2020-08-27:
Please try using the attached pyboard.py (unzip first) to run your original command with light.py.  That will print out lots of debugging, and then paste the debugging output here.

[pyboard.zip](https://github.com/micropython/micropython/files/5133784/pyboard.zip)

mridwang, 2020-08-27:
Sir, It is giving the following output on Ubuntu Terminal: 

python3 pyboard.py --device /dev/ttyACM0 light.py 
[enter 1]
[enter 2]
[a:b'\r']
b'\r'
could not enter raw repl

dpgeorge, 2020-08-27:
Please try with this pyboard.py

[pyboard.zip](https://github.com/micropython/micropython/files/5133858/pyboard.zip)

mridwang, 2020-08-27:
Sir, It is continuously printing "[c:b'r']" so many times. Following is the output. 

python3 pyboard.py --device /dev/ttyACM0 light.py
[enter 1]
[enter 2]
[a:b'r']
[c:b'r']
[c:b'r']
[c:b'r']
[c:b'r']
[c:b'r']
[c:b'r']
[c:b'r']
[c:b'r']
[c:b'r']
[c:b'r']
[c:b'r']
[c:b'r']
[c:b'r']
[c:b'r']
[c:b'r']
[c:b'r']
[c:b'r']
[c:b'r']
[c:b'r']
[c:b'r']
[c:b'r']
[c:b'r']
[c:b'r']
[c:b'r']
[c:b'r']
[c:b'r']
[c:b'r']
[c:b'r']
[c:b'r']
[c:b'r']
[c:b'r']
[c:b'r']
[c:b'r']
[c:b'r']
[c:b'r']
[c:b'r']
[c:b'r']
[c:b'r']
[c:b'r']
[c:b'r']

dpgeorge, 2020-08-27:
I assume that is a `b'\r'` (note the slash).

It looks like there is something wrong with the serial communications: either the board itself or maybe the pyserial library.  I would suggest trying a different PC.

mridwang, 2020-08-27:
Okey sir, I try to get another Ubuntu PC and will let you know the output.

dpgeorge, 2020-08-27:
You can also try Windows, or a different Ubuntu/Linux version.

mridwang, 2020-08-27:
Sir, I have tried on another PC with the same Ubuntu version "Ubuntu 20.04.1 LTS" getting the same error. 
python3 pyboard.py --device /dev/ttyACM0 test.py 
b'r'
could not enter raw repl

Note: On REPL I am able to program the device on both PC's.

May I go for another Ubuntu version?
In windows it is time consuming to install Zephyr. 

Please advise. 
Thank you.

mridwang, 2020-08-29:
Sir, I have tried Fedora as well. 
Either it is giving the following error or hangs for so long. 
python pyboard.py -c 'print(1+1)'
b'a93a378e9 on 2020-08-'
could not enter raw repl

2. Moreover, I have tried the following debug as suggested by you on n Fedora as well. 
To debug try the following:

    open the port with a serial/terminal program like picocom
    enter ctrl-C followed by ctrl-B; it should print the banner and give a >>> prompt
    enter ctrl-A; it should say raw REPL; CTRL-B to exit
    enter ctrl-D; it should say MPY: soft reboot then raw REPL; CTRL-B to exit
    enter print(1) then ctrl-D; it shou/d say OK1
    enter ctrl-B; it should print the banner and be back at the >>> prompt
3. When I press ctrl-D it does not show OK1. 

Please advise. I will be grateful.

mridwang, 2020-08-29:
Moreover sir, when I try to copy file then I get the following error. 

[mrizwan@localhost tools]$ python3 pyboard.py --device /dev/ttyACM0 test.py 
b'\r\n>>> \r\n>>> \r\n>>> \r\n>>> \r\nraw'
could not enter raw repl

mridwang, 2020-08-29:
@dpgeorge sir I got a PCA10059 dongle (want to try what response will it give) but I do not know what Configurations I am required to enter in Configuration file. I have tried of the nr52840 dk's one but it is giving error. Please advise. 

make BOARD=nrf52840dongle_nrf52840

make[3]: *** [zephyr/CMakeFiles/zephyr_prebuilt.dir/build.make:110: zephyr/zephyr_prebuilt.elf] Error 1
make[3]: Leaving directory '/home/rg/zephyrproject/micropython/ports/zephyr/outdir/nrf52840dongle_nrf52840'
make[2]: *** [CMakeFiles/Makefile2:2851: zephyr/CMakeFiles/zephyr_prebuilt.dir/all] Error 2
make[2]: Leaving directory '/home/rg/zephyrproject/micropython/ports/zephyr/outdir/nrf52840dongle_nrf52840'
make[1]: *** [Makefile:84: all] Error 2
make[1]: Leaving directory '/home/rg/zephyrproject/micropython/ports/zephyr/outdir/nrf52840dongle_nrf52840'
make: *** [Makefile:86: all] Error 2

mridwang, 2020-08-30:
@dpgeorge so far, I have tested on five nrf52840 dk's with operating systems Ubuntu, Fedora by changing the PC's as well but I am getting the same error as mentioned above. Please guide me.

dpgeorge, 2020-08-30:
> I got a PCA10059 dongle (want to try what response will it give) but I do not know what Configurations I am required to enter in Configuration file. I have tried of the nr52840 dk's one but it is giving error

The `nrf52840dongle_nrf52840` board uses `CONFIG_UART_INTERRUPT_DRIVEN=y` and that option doesn't currently work with the MicroPython configuration.

> so far, I have tested on five nrf52840 dk's with operating systems Ubuntu, Fedora by changing the PC's as well but I am getting the same error as mentioned above

Then there must be something wrong with the UART/console configuration for this nrf board.  I do not have such a board to test it with.  I have a PCA10040 and that works for me with `make BOARD=nrf52dk_nrf52832`.  I suggest you try one of those boards, or another nrf board.

mridwang, 2020-08-30:
Sir, on which operating system you have tried PCA 10040?

dpgeorge, 2020-08-30:
> on which operating system you have tried PCA 10040?

Arch Linux.

mridwang, 2020-08-30:
> Then there must be something wrong with the UART/console configuration for this nrf board. I do not have such a board to test it with. I have a PCA10040 and that works for me with `make BOARD=nrf52dk_nrf52832`. I suggest you try one of those boards, or another nrf board.
 
Thanks. My board is working fine with REPL and programing fine. The problem occurs when I want to copy python script to the board.

mridwang, 2020-08-31:
Sir @dpgeorge, please advise, is there any alternative way to copy the Py Script to nrf52840 as I want to do my work on micropython. Please advise as my work is stopped due to the issue I am facing. Hoping for your kind response. Thank you.

dpgeorge, 2020-08-31:
There seems to be two issues here:
1. pyboard.py cannot enter raw REPL
2. there is no filesystem (by default) on the nrf52840dk_nrf52840 board

These need to be fixed independently to each other.  Does pyboard.py work (to run a script, not copy files) with a standard build of the nrf52840dk_nrf52840 (ie without filesystem modifications)?

With the filesystem enabled, can you create files at the REPL?

An alternative to copying files to the device is to freeze them into the firmware by making a `scripts/` subdir in the `ports/zephyr/` directory, putting the scripts there, then rebuilding (it make take a `make clean` then `make` to pick the scripts up).

mridwang, 2020-08-31:
1. I can "make" without touching the boards configuration and making ".conf". 
Note: For nrf52840 dk, there is no default configuration file, I copy the reel_board configurations on a separate file named as "nrf52840dk_nrf52840.conf". 
2. After "make", I "west flash" and then with the help of "Putty", I can type the code on REPL and it run's on the board (example switching on the LED). 
3. I think this command is to run the script "python3 pyboard.py --device /dev/ttyACM0 test.py" which we are trying for few days. Please advise. 
>With the filesystem enabled, can you create files at the REPL?
4. Please guide me. 
>An alternative to copying files to the device is to freeze them into the firmware by making a scripts/ subdir in the ports/zephyr/ directory, putting the scripts there, then rebuilding (it make take a make clean then make to pick the scripts up).
5. I made sub dir "/scripts" in the /ports/zephyr/ directory and run the "make BOARD=nrf52840dk_nrf52840" command. But no effects on the board. 
Please guide thanks.

mridwang, 2020-09-01:
@dpgeorge sir, am I doing mistake somewhere in code building (as mentioned in my previous message)?

MaureenHelm, 2020-09-02:
> 1. When I enable the configurations CONFIG_FLASH=y, CONFIG_FLASH_MAP=y, CONFIG_MPU_ALLOW_FLASH_WRITE=y, I can make and can enter the REPL just like as mentioned in "1" without configuration. Also, I can type in the code to switch on LED etc. But is it possible to create files as micropython does not support "import os". Please guide.

What happens when you run `import os` in the REPL? Can you paste the error message?

You probably need to format the flash storage partition with a new filesystem. The following works in the REPL on the reel_board (nRF52840):

```
>>> *** Booting Zephyr OS build zephyr-v2.3.0  ***
MicroPython v1.12-700-g0e6ef4035-dirty on 2020-09-01; zephyr-reel_board with nRF52840_QIAA
Type "help()" for more information.
>>> import os
>>> from zephyr import FlashArea
>>> bdev = FlashArea(FlashArea.STORAGE, 4096)
>>> os.VfsLfs2.mkfs(bdev)
>>> os.mount(bdev, '/flash')
>>> with open('/flash/main.py','w') as f:
...     f.write(r'print("hello world!")')
... 
21
>>> print(open('/flash/main.py').read())
print("hello world!")
>>> 
```
Now if you reset (CTRL-D), the zephyr port will automatically mount the flash filesystem during boot and run `/flash/main.py`:
```
>>> 
soft reboot
hello world!
MicroPython v1.12-700-g0e6ef4035-dirty on 2020-09-01; zephyr-reel_board with nRF52840_QIAA
Type "help()" for more information.
>>>
```

mridwang, 2020-09-02:
> What happens when you run `import os` in the REPL? Can you paste the error message?
>
> You probably need to format the flash storage partition with a new filesystem. The following works in the REPL on [...]

Dear Madam, Thank you for your kind response. I am getting following error while accessing the device FlashArea. 

_Terminal ready_

._>>> import os 
.>>> from zephyr import FlashArea
.Traceback (most recent call last):
 . File "<stdin>", line 1, in <module>
.ImportError: can't import name FlashArea
.>>>_

mridwang, 2020-09-02:
> What happens when you run `import os` in the REPL? Can you paste the error message?

Dear Madam, Thank you for your kind response. I am getting the following error while accessing the device FlashArea.

Terminal ready

.>>> import os
.>>> from zephyr import FlashArea
.Traceback (most recent call last):
. File "", line 1, in
.ImportError: can't import name FlashArea
.>>>

mridwang, 2020-09-02:
>>> from zephyr import FlashArea
>>> bdev = FlashArea(FlashArea.STORAGE, 4096)
>>> os.VfsLfs2.mkfs(bdev)
>>> os.mount(bdev, '/flash')
>>> with open('/flash/main.py','w') as f:
...     f.write(r'print("hello world!")')
...     
...     
... 
21
>>> print(open('/flash/main.py').read())
print("hello world!")
>>> 
>>> 
soft reboot
hello world!
MicroPython v1.12 on 2020-09-02; zephyr-nrf52840dk_nrf52840 with nRF52840_QIAA
Type "help()" for more information.

Sorry, there was a problem with configuration file, now its running and printing hello world as shown above. So mam, is it possible to copy my own python script to the board?

mridwang, 2020-09-02:
@MaureenHelm after successfully creating file storage partition as mentioned by you, I ran the following command and same like before I cannot enter the raw REPL: 

python3 pyboard.py --device /dev/ttyACM0 light.py 
b'r'
could not enter raw repl

Please advise.

mridwang, 2020-09-05:
Dear @MaureenHelm mam and @dpgeorge sir, thank you very much for your kind help. My issue is resolved. Thank you again.

closed 2020-09-05 by mridwang

MaureenHelm, 2020-09-11:
> Dear @MaureenHelm mam and @dpgeorge sir, thank you very much for your kind help. My issue is resolved. Thank you again.

How did you resolve the raw REPL problem?

mridwang, 2020-09-13:
> > Dear @MaureenHelm mam and @dpgeorge sir, thank you very much for your kind help. My issue is resolved. Thank you again.
> 
> How did you resolve the raw REPL problem?

In reference to your last message in which you have told the methodology for creating the file system, I added the following line in place of "f.write(r'print("hello world!")')".
**f.write(r'import light')**

maweicoltsmart, 2022-08-17:
@dpgeorge I have a question about zephyr and need help, how can I compile the python script file I wrote into the firmware? Can you give an example?