micropython/micropython #6379: Add support for GT832E-01 (NRF52832) (attempt made)

JPFrancoia, 2020-08-27:
Hello,

I would like to port micropython to the GT832E-01:

[here](https://nl.aliexpress.com/item/4000110731290.html?spm=a2g0o.productlist.0.0.64c814acnsWNmg&algo_pvid=ee21218f-f71d-4ee5-b12c-3f99b6e480d3&algo_expid=ee21218f-f71d-4ee5-b12c-3f99b6e480d3-0&btsid=0b0a187b15985299349478721eacc2&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_)

This board doesn't have much documentation, but I think it's nice: it's cheap, very small, and has pins with a pitch compatible with a breadboard/perfboard.

I bought a few of them and tried to flash micropython on it. Here is what I tried:

I soldered some pins to the board and powered it. I then did a bluetooth scan and could detect a "Nordic_HRM" device -> the board is alive.

I looked at the nrf port page here: https://github.com/micropython/micropyt ... /ports/nrf, and the board I want to flash doesn't seem supported. But it's based on the NRF52832 module so I have some hope.

I modified mpconfigboard.h:

```python
#define MICROPY_HW_BOARD_NAME       "GT832E"
#define MICROPY_HW_MCU_NAME         "NRF52832"
#define MICROPY_PY_SYS_PLATFORM     "nrf52"

#define MICROPY_PY_MACHINE_UART     (1)
#define MICROPY_PY_MACHINE_HW_SPI   (1)
#define MICROPY_PY_MACHINE_TIMER    (1)
#define MICROPY_PY_MACHINE_RTCOUNTER (1)
#define MICROPY_PY_MACHINE_I2C      (1)
#define MICROPY_PY_MACHINE_ADC      (1)
#define MICROPY_PY_MACHINE_TEMP     (1)

#define MICROPY_HW_ENABLE_RNG       (1)

#define MICROPY_HW_HAS_LED          (0)

// UART config
#define MICROPY_HW_UART1_RX         (11)
#define MICROPY_HW_UART1_TX         (12)
#define MICROPY_HW_UART1_HWFC       (0)

// SPI0 config
#define MICROPY_HW_SPI0_NAME        "SPI0"
#define MICROPY_HW_SPI0_SCK         (6)
#define MICROPY_HW_SPI0_MOSI        (7)
#define MICROPY_HW_SPI0_MISO        (8)
```

I modified pins.csv:

```
ADC5,P29
ADC6,P30
ADC7,P31
GND,GND
VCC,VCC
P6,P6
P7,P7
P8,P8
P9,P9
P10,P10
RX,P11
TX,P12
P014,P14
```

I was able to successfully compile a firmware (I did run commands like `make BOARD=gt832e`), with and without the bluetooth stack (I tried to flash both, see below).

I then went on to flashing the board.

I have a STLINK-V2 connector and I connected it to the SWDIO and SWCLK pins of the board (and to VCC and GND). I installed openocd from the latest commit on github (so it has support for the nrf52 chips).

 If I run openocd, it seems my board is recognized:

```
❯ openocd -f nrf52.cfg
Open On-Chip Debugger 0.10.0+dev-01383-gd46f28c2e-dirty (2020-08-21-18:25)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
WARNING: interface/stlink-v2.cfg is deprecated, please switch to interface/stlink.cfg
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD

nRF52 device has a CTRL-AP dedicated to recover the device from AP lock.
A high level adapter (like a ST-Link) you are currently using cannot access
the CTRL-AP so 'nrf52_recover' command will not work.
Do not enable UICR APPROTECT.

Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : clock speed 1000 kHz
Info : STLINK V2J29S7 (API v2) VID:PID 0483:3748
Info : Target voltage: 3.174546
Info : nrf52.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : starting gdb server for nrf52.cpu on 3333
Info : Listening on port 3333 for gdb connections
```

I then flashed the firmware that I just built:

```
❯ telnet localhost 4444                                  
Trying ::1...
Connection failed: Connexion refusée
Trying 127.0.0.1...
Connected to localhost.
Escape character is '^]'.
Open On-Chip Debugger
> program firmware.hex       
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x00015908 msp: 0x20010000
** Programming Started **
nRF52832-QFAA(build code: E0) 512kB Flash, 64kB RAM
Adding extra erase range, 0x00023bfc .. 0x00023fff
** Programming Finished **
> program firmware.hex verify
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x00015908 msp: 0x20010000
** Programming Started **
Adding extra erase range, 0x00023bfc .. 0x00023fff
** Programming Finished **
** Verify Started **
** Verified OK **
```

It seemed all right.

I then disconnected the STLINK connector. I connected the tx/rx pins (11 and 12) to a FTDI adapater, and powered the board. I tried to access the micropython prompt with:

```
screen /dev/ttyUSB0 115200
```

But I could never get my hands a prompt (seems like communication between my PC and the board doesn't happen). If I do a bluetooth scan when the board is powered, I do not see a Nordic device anymore, which I assume means I wrote *something* to the board.

Could you please point me in the right direction?

Do you see anything weird in the things I did?

I'll be happy to write an article or make a PR if I manage to get this working.

JPFrancoia, 2020-09-11:
UPDATE: I kept trying to get this to work and I think I got somewhere.

I ended up flashing the board with pyocd, like this:

```
make V=1 BOARD=gt832e SD=s132 sd FLASHER=pyocd flash
```

Now when I connect the tr/rx pins to FTDI adapter, and monitor the serial connection output on my computer, I get nothing *at first*. But when I hot unplug/plug the FTDI adapter into the breadboard, I finally get a prompt! The FTDI adapter needs to stay connected to my computer. The board stays connected to the breadboard. Unpluging/pluging the FTDI adapter into the breadboard magically gets me a prompt.

Everything works as expected after that: I can import the ubluepy module and do a bluetooth scan.

Do you have any idea what's happening with the prompt at first? It looks like I'm maybe missing a connection with the board. So far I have:

Tx -> Rx of FTDI adapter
Rx -> Tx of FTDI adapter
Vcc -> 3.3 V
GND -> GND

Any help is appreciated.

jonnor, 2025-10-03:
This seems possible to do with minor modifications to the core build, and be a relatively uncommon board. Proposing to close.